esphome/feature-requests #1276: Bring back 'Auto Save'

3ative, 2021-06-23:
**Describe the problem you have/What new integration you would like**

In the previous build (v1.18) when you clicked, what was then the "Compile' Button - now "Install", ESPHome would first 'Save' the code in the editor before taking action. 

[image]

- _It wold be great and a time saver to have this feature back._  👍

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

jesserockz, 2021-06-26:
This was a bug introduced with the dashboard changes, but has been fixed in v1.19.4

3ative, 2021-06-26:
@jesserockz 
💕That's great news, thanks for the info.

3ative, 2021-06-26:
Confirmed - Closing this thread.
Thanks to everyone involved. 👍

closed 2021-06-26 by 3ative