esphome/feature-requests #1279: Allow Secrets to Create configuration

3ative, 2021-06-27:
**Describe the problem you have/What new integration you would like**

Would it be possible to add a check to the "Create configuration" backend to allow for **Secrets** to be used?

For Example:
[image]

Currently, it results in plain text being added to the config:
```yaml
wifi:
  ssid: "!secret ssid"
  password: "!secret wifi_password"
```
- Note the included quotation marks.

Obviously, it needs to be:
```yaml
wifi:
  ssid: !secret ssid
  password: !secret wifi_password
```

**Please describe your use case for this integration and alternatives you've tried:**

It's obviously an advantage to use secrets in ESPHome and I think this feature could make the user experience smother. 

**Additional context**

Thanks. 👍

zuidwijk, 2021-06-30:
+1 from me! 

Maybe a button to use default specified secrets for wifi?
[image]

3ative, 2021-06-30:
An extra Button could/would be a nice option too.