esphome/feature-requests #1281: Grouping?

atrent, 2021-06-28:
**Describe the problem you have/What new integration you would like**

1) I define a "complex" (with many subcomponents) device in a .yaml file, with homeassistant integration (i.e., api:)
2) I flash it on my physical device
3) I let it discover by my homeassistant instance and I "integrate" it
4) my "complex" device appears scattered on the "overview" page of homeassistant, i.e., every subcomponent in a different group (e.g., Sensor, BinarySensor, Switch, etc.)

I would like to see my complex device as a single tab on the overview page, I searched the wiki/issues/etc. but I did not find a clue, feel free to RTFM me

**Please describe your use case for this integration and alternatives you've tried:**

I'm defining many complex objects using esphome and I'd like to avoid re-grouping them editing yamls on homeassistant

**Additional context**

Thanks

atrent, 2021-06-30:
Self-reply: the autodiscovered device can be added (from Configuration -> Devices) to lovelace and stays grouped, just need to give it a name... I hope there's a way to specify a Device Name in the esphome .yaml

nagyrobi, 2022-07-07:
Please head on to Discord channel for support questions.

closed 2022-07-07 by nagyrobi