esphome/feature-requests #1284: Adds climate - ir - electrolux

Dilkins85, 2021-06-30:
Please add control of the electrolux air conditioner via infrared port with feedback, since my air conditioner does not have a uart port (((and I want to integrate it somehow. I tried to integrate it through a whirlpool, it works but the feedback is not stable and the temperature does not match

I tried to integrate via whirlpool, the feedback is not stable (and the temperature does not match

**Additional context**

kevinblo, 2021-06-30:
Yes, yes, yes!

Dilkins85, 2021-07-16:
empirically I found out that the whirlpool component is suitable for my electrolux, but there is one discrepancy, if I control the air conditioner from the remote control, then the temperature in the home assistant is thrown correctly (control panel-24, ON-24, air conditioner-24), and if I send a command from ON , let's say 24 degrees, then the air conditioner turns on at 22 degrees (. Can I somehow fix this?

Dilkins85, 2021-09-02:
I would still like to make such an integration)))