esphome/feature-requests #1286: Integrate disabling deep sleep for OTA updates into the deep_sleep component

KeyserSoSay0, 2021-07-01:
**Describe the problem you have/What new integration you would like**

Basically I want to be able to do something like 
deep_sleep:
  id: deep_sleep_control
  sleep_duration: 5m
  disable_entity: input_boolean.disable_deep_sleep

where input_boolean.disable_deep_sleep would be a boolean coming from the home assistant dashboard.

**Please describe your use case for this integration and alternatives you've tried:**

Currently, you can do this by adding some yaml code, but it's kind of a pain.  I don't know that this would normally be worth doing, but there's also another feature request to make it so that devices connect to a central server to access the hassio api instead being the server themselves.  If the disable deep sleep boolean were integrated into the component, then the component could make sure to not go back to sleep until it grabbed the up to date value from the server after connecting.  I imagine that's how 99% of people would want to do it in yaml anyways, so why not just integrate it directly to save them the trouble (unless I'm missing something.  I'm just getting started on ESPHome).

**Additional context**

KeyserSoSay0, 2021-07-01:
Actually, I was thinking about it more and you could probably fully integrate OTA updates and deep_sleep into the new API central server stuff.  Basically, when you click to do an OTA update, it could add that device to a list of devices that want to update.  Then, whenever a device connects to the server, the server checks to see if that device is on the update list.  If it is, then it tells the device to disable deep_sleep so that it can get an update.

If you really wanted to, you could probably even save the update payload on the server so that users could basically click the OTA update button and know it would happen eventually.

elupus, 2022-01-18:
I think it can be generalized to a deep sleep condition that is evaluated before sleep. And have the ota component expose its state. then it should be rather simple to skip a deep sleep interval with a simple condition (could also support other conditions to skip the sleep).