esphome/feature-requests #1287: Add suport: Seeed Grove Multichannel Gas Sensor v2 (CO, NO2, C2H5CH and VOC)

vkcvkcvkc, 2021-07-01:
**Describe the problem you have/What new integration you would like**

It would be nice to add support for this multi-gas sensor from Seeed Studios with I2C interface.
Four fully independent sensor elements on one package, ability to detect Carbon monoxide (CO), Nitrogen dioxide (NO2), Ethyl alcohol (C2H5CH) and Volatile Organic Compounds (VOC).

**Please describe your use case for this integration and alternatives you've tried:**

This multi-gas sensor would be a very nice addition.

**Additional context**

Link to the sensor : https://wiki.seeedstudio.com/Grove-Multichannel-Gas-Sensor-V2/
https://www.seeedstudio.com/Grove-Multichannel-Gas-Sensor-v2-p-4569.html
https://eu.mouser.com/ProductDetail/713-101020820

Library: https://github.com/Seeed-Studio/Seeed_Multichannel_Gas_Sensor/archive/master.zip

asucrews, 2022-03-11:
@vkcvkcvkc does this help https://github.com/MatejMecka/grove-multichannelgassensor-esphome

asucrews, 2023-06-05:
i think this request is being filled by this PR https://github.com/esphome/esphome/pull/4304