esphome/feature-requests #1289: Rename git master branch to main

OttoWinter, 2021-07-02:
GitHub is changing the default branch naming from `master` to `main`, and I support that.

In ESPHome we still use `master` in a bunch of places, I'd like to rename those. Specifically:
- [esphome](https://github.com/esphome/esphome)
  -	master :arrow_right: release (`release` is better IMO as it makes it clear that branch is only for releases, and we don't get accidental PRs against it)
- [esphome-vscode](https://github.com/esphome/esphome-vscode)
  -	master :arrow_right: main
- [hassio](https://github.com/esphome/hassio)
  -	master :arrow_right: main
- [esphome-release](https://github.com/esphome/esphome-release)
  -	master :arrow_right: main
- [esphome-docker-base](https://github.com/esphome/esphome-docker-base)
  -	master :arrow_right: main
- [issues](https://github.com/esphome/issues)
  -	master :arrow_right: main
- [feature-requests](https://github.com/esphome/feature-requests)
  -	master :arrow_right: main
- [probot-esphome](https://github.com/esphome/probot-esphome)
  - master :arrow_right: main

I already trialed this process with [aioesphomeapi](https://github.com/esphome/aioesphomeapi/), and it went quite smoothly. The GitHub frontend even shows a message with instructions for people who already cloned it. More on the process here: https://github.com/github/renaming

Todo:
 - [ ] Update esphome-release for new naming
 - [ ] Update probot-esphome `merging-to-master` system
 - [ ] Update GitHub Actions workflow files
 - [ ] Rename the branches

Impact: I think not big. Existing PRs against esphome/esphome-docs are not impacted in any way. Only the other projects, but those have very few pull requests created anyway.

CC @jesserockz & @glmnet (for esphome-vscode)

jesserockz, 2021-07-02:
I have done this on a few (personal and other) repos and can only say it went smoothly.

Makes it easier when we have dev/current/next where prs go to.

OttoWinter, 2021-07-02:
Ok, everything should be updated now.

I only thought of it afterwards, but I hope ~~Hass.IO~~ Home Assistant add-on repo checker won't freak out due to the changed name. I hope not, but we'll see :crossed_fingers:

closed 2021-07-02 by OttoWinter