esphome/feature-requests #1291: I need Edge-triggered Switch or Binary_Sensor

erreipi, 2021-07-03:
**Describe the problem you have/What new integration you would like**

Currently, the switch or binary_sensor status updates during the on period or off period, it is called level-triggering.
But, I need to design an edge detection as the switch or binary_sensor status updates during (On -> Off) or (Off -> On) period.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

Ex)
Level-triggered Switch : on_turn_on:  or on_turn_off:
Edge-triggered Switch : on_posedge:  or on_negedge:

bkaufx, 2021-07-07:
on_turn_on and on_turn_off are edge triggering.

bkaufx, 2021-07-07:
Same with on_press and on_release for binary sensors.

erreipi, 2021-07-08:
> on_turn_on and on_turn_off are edge triggering.
Oh, there is a way. Thank you !!!

closed 2021-07-08 by erreipi