esphome/feature-requests #1294: name_add_mac_suffix  for hotspot name

vzagorovskiy, 2021-07-05:
**Describe the problem you have/What new integration you would like**
Add name_add_mac_suffix  option for ssid for “Access Point Mode”

```yaml
wifi:
  ap:
    ssid: "Fallback Hotspot"
    password: "W1PBGyrokfLz"
    name_add_mac_suffix: true
```

**Please describe your use case for this integration and alternatives you've tried:**
Configuring multiple devices at the same time that use single config with name_add_mac_suffix option 

**Additional context**

This option will greatly improve usability when using a single config for many devices.

vzagorovskiy, 2021-07-22:
Added in #1990

closed 2021-07-22 by vzagorovskiy

raidnet-ms, 2021-07-23:
is the dashboard now supporting macsuffix for OTA?

vzagorovskiy, 2021-07-23:
> is the dashboard now supporting macsuffix for OTA?

No. Dashboard doesn't know the node address.

> Using name_add_mac_suffix allows the user to compile a single binary file to flash many of the same device and they will all have unique names/hostnames. Note that you will still need to create an individual YAML config file if you want to OTA update the devices in the future.

raidnet-ms, 2021-07-23:
Well even if i created an individual yaml file I was unable to see the device online into the dashboard.

vzagorovskiy, 2021-07-23:
If you have created individual yaml file you should not use "name_add_mac_suffix", you should use a node name e.g. MyNode-E01334 or use "manual_ip" for this node. "name_add_mac_suffix" option is only useful if you need to make one firmware for multiple  devices and give them different names.