esphome/feature-requests #1297: Implement "select" entity from Home Assistant

jhenkens, 2021-07-09:
**Describe the problem you have/What new integration you would like**

HomeAssistant recently released [select entities](https://www.home-assistant.io/blog/2021/07/07/release-20217/#new-entity-select) which I think have some valid use cases in mapping ESP devices. 
**Please describe your use case for this integration and alternatives you've tried:**

Using multiple internal relays and exposing one "drop down" would works similarly to exposing the relays themselves and using interlock groups, except it is more explicit in its function once exposed to the outside world.

**Additional context**

cliffordwhansen, 2021-07-16:
To add to the use cases:
- ESP Powered airfreshner, can expose a dropdown with pre set time intervals (9, 18, 32)
- Fan can return a drop down of speeds instead of a slider

oxan, 2021-08-02:
This has landed in dev now, and will be released with v1.21.

closed 2021-08-02 by oxan