esphome/feature-requests #1298: SPEC-SENSOR CO SENSOR (ULPSM-CO 968-001)

shubness, 2021-07-09:
**Describe the problem you have/What new integration you would like**
Looking to create an integration with the[ ULPSM-CO 968-001] (https://github.com/esphome/feature-requests/files/6790182/ULPSM-CO-968-001.pdf)
) 

**Please describe your use case for this integration and alternatives you've tried:**
Indoor air quality measurement - focused on carbon monoxide. 

**Additional context**
This is an analogue sensor so will require some interpretation of the functions provided in the data sheet.

shubness, 2021-07-09:
Some more context: https://github.com/SPEC-Sensors/ULPSM

loongyh, 2021-07-09:
[datasheet](https://www.spec-sensors.com/wp-content/uploads/2016/10/ULPSM-CO-968-001.pdf)

```yaml
sensor:
  ulpsm:
    vgas: GPIO* (Required)
    vtemp: GPIO* (Required)
    voffset: float (Required)
    sensitivity_code: float (Required)
    gas: one of [co, h2s, no2, so2, o3, c2h5oh, iaq, ri] (Required)
```

shubness, 2021-07-16:
@loongyh any update on your efforts?

loongyh, 2021-07-16:
Still puzzled on temperature compensation, pg. 4 of the datasheet sets the reference point at 20°C for zero shift and span, there's a table stating the %/ppm to increment/decrement per °C, however it did not state how to derive a °C reading from the vtemp voltage.

Meanwhile can you test this on your setup?

```yaml
globals:
  - id: voffset
    type: float
    initial_value: '0.003'  # Replace with your sensor's bias (1mV = 0.001)
  - id: sensitivity_code
    type: float
    initial_value: '4.41'  # Replace with your sensor's sensitivity code
  - id: tia_gain
    type: uint16_t
    initial_value: '100'  # For CO gas
  - id: vgas0
    type: float
  - id: m
    type: float

i2c:
  sda: 21  # Replace with your sda pin
  scl: 22  # Replace with your scl pin

ads1115:
  - address: 0x48  # Replace with your ADS1115's i2c address

sensor:
  - platform: ads1115
    multiplexer: 'A0_GND'
    gain: 4.096
    id: vgas
  - platform: ads1115
    multiplexer: 'A1_GND'
    gain: 4.096
    id: vtemp
  - platform: ads1115
    multiplexer: 'A2_GND'
    gain: 4.096
    id: vref
  - platform: template
    name: "ULPSM"
    lambda: |-
      id(vgas0) = id(vref).state + id(voffset);
      id(m) = id(sensitivity_code) * id(tia_gain) * 0.000001;
      return (1 / id(m) * (id(vgas).state - id(vgas0)));
```

shubness, 2021-07-16:
@loongyh thanks for the feedback. I'm a bit confused about vref- do I use the voltage ladder as per the diagram or do I wire it to an ADC input? As per the code, it looks like it goes to an ADC? How does this work?

loongyh, 2021-07-16:
Try this:

[image]

shubness, 2021-07-16:
@loongyh I'm getting a reading with the above set up of **1,059.0 ppm** - which is not possible.

shubness, 2021-07-16:
snapshot of the ppm reading i'm getting over +-20 min
[image]

loongyh, 2021-07-22:
vgas at zero concentration should be Vcc/2 = 3.3/2 = 1.65V. Can you use a voltmeter to check if your vgas reads 2.128V?

closed 2022-07-01 by nagyrobi