esphome/feature-requests #1302: Climate IR LG - Support fan only modes and all "on" commands

danieldabate, 2021-07-13:
**Describe the problem you have/What new integration you would like**

I have an LG central air conditioner unit with a PREMTB00 wall remote control. I found that there are a couple of things not currently supported by the existing implementation:
1. Fan only mode
2. "On" commands to start the air conditioner in a mode different than cool

**Please describe your use case for this integration and alternatives you've tried:**

I created for my specific use case an external component cloning and modifying the current one. I'd like to share my changes and create a pull request, but I don't want to brake existing use cases.

Do all LG units support fan only mode? If not, I was thinking about making it available in the config as a parameter, but considering the value is passed in the constructor of the parent class I reached a dead end. Not proficient on C++ at all.

Any help is greatly appreciated.

**Additional context**

No additional comment

danieldabate, 2022-05-22:
I uploaded the changes to my fork in case anyone is willing to test it:
https://github.com/danieldabate/esphome/tree/dev

closed 2022-07-07 by nagyrobi

tomeeka, 2022-08-11:
working great as an external_component!
thanks for sharing!

danieldabate, 2022-08-11:
Thanks for testing! 

I'll submit a pull request with the changes during the next few days so we don’t have to use the code as an external component.

danieldabate, 2022-09-26:
> @danieldabate was it pushed already? That issue keeps me still on Tasmota. Switching from cool to heat take ages

It's pushed to my personal branch, but has not yet been merged to the main repo. So we don't have an official release with this improvement. This is the pull request I submitted: https://github.com/esphome/esphome/pull/3712

If you want to have it asap, you can try it out by putting the [climate_ir_lg folder](https://github.com/danieldabate/esphome/tree/dev/esphome/components/climate_ir_lg) in the external_component directory.