esphome/feature-requests #1306: ESPHome dashboard colors

Depechie, 2021-07-15:
**Describe the problem you have/What new integration you would like**
The current ESPHome dashboard uses colors to indicate if a device is online or not, with green and red.
But for me as a person with problems seeing the difference with colors this is not helpfull.

Can icons and/or text also be used to give an indication of the status?

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

koenvervloesem, 2021-07-15:
Yes, there's already an issue https://github.com/esphome/issues/issues/1504 and feature request https://github.com/esphome/feature-requests/issues/890 for this.

In the mean time, maybe this is helpful: a commenter in the latter feature request suggests to use the browser extension Dark Reader because with a dark background the difference should be easier to see.

Depechie, 2021-07-15:
thx for the links @koenvervloesem ! Means this can be closed....

closed 2021-07-15 by Depechie