esphome/feature-requests #1308: GM65 1D 2D Code Scanner Bar and  QR Code Reader

Dodoooh, 2021-07-20:
**Describe the problem you have/What new integration you would like**
I found this awesome barcode reader: 
[M65 1D 2D Code Scanner Bar Code Reader QR Code Reader Modul ](https://a.aliexpress.com/_mLZvjgT
)

which can be connected via USB as well as UART. 
Unfortunately I don't have one at hand yet, I just ordered one, but it would be great to include it as a standard sensor in ESPHome.

**Please describe your use case for this integration and alternatives you've tried:**
I think there are many ways to use QR code scanner in Home Assistant.

few examples:  
- sending friends a QR code which will open their front door.
- filling up shopping lists by scanning the empty packages

**Additional context**

pove, 2021-10-01:
Very interesting as 2D scanner as well. I can imagine a project to read all your shopping items at home to add them automatically to your Grocy Erp inventory.

Did you make some progress on this?

pove, 2021-10-01:
I asume you will need a USB TTL adapter or a level converter/shifter (5V - 3.3V), right?

nagyrobi, 2022-07-06:
See https://esphome.io/cookbook/uart_text_sensor.html

IgorDeryabkin, 2024-04-03:
Good afternoon!
The Grow GM65 scanner is not working correctly.
PDF417 code in HEX1: 33 65 65 7A 33 7A 65 65 65 65 30 7A 65 34 33 69 62 7A 39 7A 7A 30 39 65 65 65 6A 7A 65 74 74 7A 30 6A 74 7A 74 74 33 65 6A 74 6A 30 7A 74 65 7A 16 B7 1F 0C 00 00 00 00 00 00 00 00 0C 78 06 66 3C 77 3E 00 7C 19 03 10 4F 3D 23 3C 2E AC 47 1A 90 04 4C AC AD 45 37 44 3 E 24 46 90 0C 56 1A 13 29 68 1C 1B 4C 5C 55 A9 20 46 4D 32 54 3D 0B 10 37 19 AD 0B 49 54 2D 19 26 5A 56 B7 0C 0A 12 BB 5B 57 46 4C 2F 2E 27 0D

Grow GM65 reads HEX1: 33 65 65 7A 33 7A 65 65 65 65 30 7A 65 34 33 69 62 7A 39 7A 7A 30 39 65 65 65 6A 7A 65 74 74 7A 30 6A 74 7A 74 7 4 33 65 6A 74 6A 30 7A 74 65 7A 16 00 00 0C 00 00 00 00 00 00 00 00 0C 78 06 66 3C 77 3E 00 7C 00 00 00 00 00 00 2E 00 00 1A 01 05 4C 00 00 24 46 01 AD 56 00 00 29 68 00 00 4D 00 00 0B 00 00 00 00 A6 19 00 00 00 00 5A 56 00 00 00 00 00 00 00 00 4C 2F 00 00 0D

PDF417 example attached.
Help me fix the error!
[image]