esphome/feature-requests #1310: Wifi Switch

edunhill, 2021-07-22:
**Describe the problem you have/What new integration you would like**
Ability to switch wifi chip on and off at will

**Please describe your use case for this integration and alternatives you've tried:**
Primary use case is to toggle wifi with a hardware switch to save battery. Would also like to toggle wifi in code when for eg a condition is met to update MQTT/HA (again to save battery)

Have investigated use of deep sleep but other functions are still required when wifi is not therefore doesn't suit my use case.

Have tried below (undocumented lambda) which does seem to take wifi down briefly but guess API is bringing it back up automatically.
- lambda: WiFi.mode(WIFI_OFF);
- lambda: WiFi.mode(WIFI_STA);

**Additional context**

rmesnard, 2021-07-28:
Yes , i need also this functionality. Use cases for me :

1. Use the power of ESP and ESPHome to make easy some simple automation system. Disable Wifi to lighten the home wifi network. Switch to enable Wifi only when update required.
2. For more security purpose , use of ESP32 Ethernet instead of Wifi in  Access control or Alarm system .

edunhill, 2021-07-28:
On your second point if you ommit the wifi section from your yaml then wifi is disabled - I use this with ethernet based ESP32 such as the Olimex POE unit. I'm not 100% sure it fully disables the chip though.

VKlueber, 2021-07-31:
I also need this:
I want to use ESP Home nodes to control the shades everywhere in our house. We don´t like the idea of having ~ 20 nodes active on WiFi around-the-clock.
That´s why I use the CAN Bus option to communicate with the nodes controlling the shades. I only need WiFi in case I need to to OTA because I can not access them as they are mounted in the walls of my house. In that cases I want to switch WiFi `on`/`off`via CAN Bus.

VKlueber, 2021-07-31:
see https://github.com/esphome/feature-requests/issues/1236, maybe this simply can be merged?

jsantosdias, 2023-05-01:
For me it would also be especially useful! I just want a light detector for mailbox, but uses batteries... If WiFi is always on, batteries last 2 days. If I was able to turn Wifi off and only on when adc goes above some number, and then shut down again and sleep, I think it would work. I have tried Deep Sleep, but it still communicates with HASS while sleeping...

DJTerentjev, 2023-10-18:
Yes , i need also this functionality.