esphome/feature-requests #1314: Add entity_id as sensor option

JeeCee1, 2021-07-25:
**Describe the problem you have/What new integration you would like**

To prevent having to manually rename the friendly_name of entities in HA it would be very convenient to have this option directly in ESPHome. A nce description has been given [here](https://github.com/home-assistant/core/issues/35683#issuecomment-735252951) and [here](https://github.com/home-assistant/core/issues/47546).
**Please describe your use case for this integration and alternatives you've tried:**

The benefit is that you still have a very detailed, descriptive (possibly unreadable) entity_id in HA, enabling to deduce where a sensor is located and originating from (eg. by prepending `esp`) while still having nice readable names.

Of course this can be done in `customisation.yaml` in HA, but it's more neat to do this at the source: in the ESPHome yaml.
**Additional context**

fede843, 2021-11-18:
I would like the flexibility of having names and ids translated from sensors id and names. Currently is using names for both, and it is an inconvenience in some situations.

rrooggiieerr, 2022-03-29:
Totally agree that it's very inconvenient to have the HA Entity ID's based on the ESPHome Entity Names

closed 2022-07-01 by nagyrobi