esphome/feature-requests #1317: "Update selected" option to accompany Update All

DorjeDorf, 2021-07-27:
**Describe the problem you have/What new integration you would like**

I don't always want to update all of the devices at the same time (especially in case some power devices get reset and turn off the plugs). 

**Please describe your use case for this integration and alternatives you've tried:**

Imagine you have devices like a dishwasher/fridge/any other device drawing current. You have that connected to a power plug. You run an update - the plug resets cutting power off from the device (even if for just a second until it turns back on - depending on it's resume power setting). 

Right now, I'm just manually going item by item with devices I know can be upgraded together and pick the right time to upgrade those critical ones. 

I would like a checklist to become available where I can chose which multiple devices can get updated in bulk. 

**Additional context**

ariedeleen, 2021-08-01:
@DorjeDorf update selected would be a nice beneficial feature.
And also it's starts all over again, updating all... Even if there are devices already updated or not connected?
This is also time consuming. With update selected it would be possible to skip these in advance.

DorjeDorf, 2021-08-02:
@ariedeleen - well here my approach would be a typical software dev one ;) 

If you don't need something updated, because it's already on the newest build - don't select it when picking items for batch update (here I would love an icon on the selection list that would show if the item needs a version bump or not - just like the cards have it). 

Also, since this is a batch process - you don't really care how much time it takes - you just set it and forget it for a while, while it chugs along upgrading things in the background tab (or even better without a tab open). 

I can see another scenario, where imagine you want to change the code of a number of devices because you made a mistake when coding them (totally never happened to me! :) ) or found a cool new way of implementing something you'd like to use. Then, you'd just update the code and then instead of hitting install now - either hit a new button Queue or save the code and then select these items for batch upgrade.

mkfink, 2021-08-05:
An additional use case for me is when I have a few dozen new devices to serial flash for the first time, it's far more convenient to build everything first. That way I either end up with a folder full of *.bin files or else just all the build files are generated, so that the next time I build from the UI it doesn't have much to compile. My current workaround is this, which works pretty well, but I still have to do some juggling of files or creative globbing to get just the list of files I want to compile:
```
import glob
from subprocess import call

# get a list of yaml files, either from glob or some other way
file_list = glob.glob('./*.yaml')

for file in file_list:
    print(file)
    devname = file[0:-5] # strip off '.yaml'
    build_cmd = f'esphome {file} compile'
    call(build_cmd, shell=True)
    cp_cmd = f'cp build/{devname}/.pioenvs/{devname}/firmware.bin ./firmwares/{devname}.bin'
    call(cp_cmd, shell=True)
```

davidclemens, 2021-08-21:
+1
I'd love this feature in the dashboard as well.

UnePierre, 2023-04-26:
> An additional use case for me is when I have a few dozen new devices to serial flash for the first time, [...]
Maybe use the [name_add_mac_suffix](https://esphome.io/components/esphome.html#esphome-mac-suffix) feature? I deploy one binary on tens of devices and only later add them upon installation. Works quite well.

But I also would like to "update until all are up-to-date" or "update selected" or similar, because on my currently only 24 devices, updating all isn't stable. Also, I'd like to avoid the really non-necessary flash cycles of a repeated "update all" loop until all devices are updated.

ariedeleen, 2023-04-28:
Also mentioned a select option for the ESPHome HA integration. On the [Home Assistant community website](https://community.home-assistant.io/t/bulk-update-devices-feature-esphome-in-ha-settings/538698)

randybb, 2023-05-01:
It makes more sense to offload all such features to HA as in HA it is is pretty easy to create groups of devices, create automations,...

pmannk, 2023-05-01:
> It makes more sense to offload all such features to HA as in HA it is is pretty easy to create groups of devices, create automations,...

Many people run ESPHome without HA, or on a separate host/instance to their HA installation, so I think it's quite useful to have this feature in ESPHome too, not just specific to HA.

robsonfelix, 2023-07-12:
Any ETA on this? I have a big list of devices that keeps growing and updating them is becoming extremely painful.

JonLaliberte, 2025-04-16:
I run into a problem where the update all updates a device that powers off my router, so then none of the other updates work. Maybe I can prefix it with "Zzzzz" to make it go last for now 😂

UnePierre, 2025-07-10:
Here is my automatic update solution, as automation in HA:

It takes upto two hours and tries every minute to update one of the online ESPhome devices with a pending update.

Let's me sleep more comfortably.

```
alias: "ESPHome: auto-update devices"
description: >-
  Find all ESP devices, filter by online and needing an update, do one at a
  time.
triggers:
  - trigger: time_pattern
    hours: "12"
    minutes: "*"
    enabled: true
  - trigger: time_pattern
    hours: "13"
    minutes: "*"
    enabled: true
actions:
  - variables:
      outdated_esp_items: |-
        {{ states.update
          | selectattr('state', 'eq', 'on')
          | selectattr('attributes.title', 'match', 'ESPHome')
          | selectattr('attributes.auto_update', 'eq', false)
          | selectattr('attributes.in_progress', 'eq', false)
          | sort(attribute='attributes.installed_version')
          | map(attribute='entity_id')
          | list }}
  - condition: template
    value_template: "{{ outdated_esp_items | count > 0 }}"
  - action: update.install
    target:
      entity_id: "{{ outdated_esp_items | first }}"
mode: single
```