esphome/feature-requests #1321: SPI Support for BMP280

psbankar, 2021-07-28:
**Describe the problem you have/What new integration you would like**

BMP280 sensor supports I2C as well as SPI but ESPHome currently supports only I2C for it. It would be great if SPI support was added for this sensor as well.

**Please describe your use case for this integration and alternatives you've tried:**

Same as above

**Additional context**

Same as above

njbuch, 2022-01-13:
I will support this feature request! Thanks

The case is that I have a need for greater distance and the sensor supports this, in the I2C case shorter distance is required.

DamianBroj, 2022-08-10:
any progress?

Stef33fr, 2022-11-16:
Did something done for this ?

IGitYou, 2023-08-31:
I vote for this for sensor distance. It was not clear to me it did not support it until I read this and I was attempting to research how to do exactly what is requested.

apbodrov, 2023-09-20:
I will try to make a version of bme280 for SPI within a couple of weeks

rossens, 2023-12-16:
I also look forward to this. 
Another advantage for SPI over I2C - more sensors can be connected.
My understanding is Pi Pico can only support 2 x I2C BME280 (unless including a multiplexer).
But with SPI can support many more sensors.

bartoszwalicki, 2024-05-17:
Is there any work in progress on this issue?

apbodrov, 2024-06-09:
> Is there any work in progress on this issue?
Hello! I don’t have a BMP280, I can try to write a pull request blindly, but it’s not very convenient

grahen, 2024-06-10:
Ja det verkar faktiskt så,

🎉

Vad är det som har åtgärdats?

Då kommer jag cancellera utbytet av min mätare =).

Med vänliga hälsningar
Henrik

IGitYou, 2024-07-11:
I happen to have one I could spare but am in the domestic US, and would not even begin to know how to get something like that to Moscow. Sorry.

ademuri, 2024-07-18:
I have a [draft pull request](https://github.com/esphome/esphome/pull/7053) which adds support for BMP280 via SPI. You can test it out by adding the following to your yaml file: 

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/ademuri/esphome
      ref: bmp280-spi
    components: [ bmp280_spi, bmp280_base ]

sensor:
  - platform: bmp280_spi
    temperature:
      name: "Board Temperature"
      oversampling: 16x
    pressure:
      name: "Board Pressure"
      oversampling: 16x
    update_interval: 5s
    cs_pin: GPIO16

spi:
  clk_pin: GPIO13
  mosi_pin: GPIO15
  miso_pin: GPIO14
  interface: hardware
```

If you do, feedback on whether it worked or not would be helpful.

closed 2024-09-05 by jesserockz

IGitYou, 2024-09-12:
Thanks all