esphome/feature-requests #1324: PN532 + TCA9548A

TheEris, 2021-07-29:
**Describe the problem you have/What new integration you would like**

Would love to have multiple RFID readers (PN532) on single ESP without the need for SS pin (PN532 with SPI).
There is already support for TCA9548A I2C multiplexer, so you can have 8 x PN532 with just two wires

**Please describe your use case for this integration and alternatives you've tried:**

There is the alternative of SPI, but you still need to have a dedicated SS pin for each PN532. With I2C you need just two wires and can have 64x PN532 on single bus (8x TCA9548A) 

**Additional context**

OttoWinter, 2021-08-01:
PN532 communication over i2c already exists: https://esphome.io/components/binary_sensor/pn532.html#over-i2c

closed 2021-08-01 by OttoWinter

TheEris, 2021-08-01:
@OttoWinter Not what I meant. I wanted to ask about support for the TCA9548A I2C multiplexer + PN532
I know the PN532 already works over i2c, that's why I'm if its possible to have multiple of them over i2c with the multiplexer, as now it doesn't seems to be working

```
substitutions:
  devicename: test
  devicename_full: test

esphome:
  name: $devicename
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "wifi"
  password: "wifipassword"

logger:

api:

ota:

i2c:
  sda: 21
  scl: 22
  scan: true
  id: i2c0

tca9548a:
  - address: 0x70
    id: multiplex0
    i2c_id: i2c0
    scan: true

pn532_i2c:
  update_interval: 1s

binary_sensor:
  - platform: pn532
    multiplexer:
      id: multiplex0
      channel: 0
    uid: 74-10-37-94
    name: "PN532 NFC Tag"
```

for instance, this trows an error `[multiplexer] is an invalid option [binary_sensor.pn532]. Please check the indentation.`
Am I just stupid and have some syntax error somewhere?

OttoWinter, 2021-08-01:
I see. Move the multiplexer block to the pn532_i2c section

TheEris, 2021-08-01:
Compiled without a problem!
Thanks!