esphome/feature-requests #1326: Mac address info for ethernet component

r312645, 2021-07-31:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2022-03-28:
https://github.com/esphome/feature-requests/issues/1683

pepsonEL, 2023-03-02:
Any news about this ? Is any chance to show MAC for Ethernet module ?

ptr727, 2024-05-29:
Looks like the ethernet component is not exposing the MAC address, so non trivial to get it.

MAC address is logged but not exposed:  
https://github.com/esphome/esphome/blob/dev/esphome/components/ethernet/ethernet_component.cpp#L504

```cpp
  uint8_t mac[6];
  err = esp_eth_ioctl(this->eth_handle_, ETH_CMD_G_MAC_ADDR, &mac);
  ESPHL_ERROR_CHECK(err, "ETH_CMD_G_MAC error");
  ESP_LOGCONFIG(TAG, "  MAC Address: %02X:%02X:%02X:%02X:%02X:%02X", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
```

`eth_handle_` is private so not accessible from a lambda.

closed 2024-06-04 by jesserockz