esphome/feature-requests #1327: Default value/min/max for number component

Weissnix4711, 2021-07-31:
**Describe the problem you have/What new integration you would like**

The number component does not currently have a way to set visual settings, such as a default number, or min/max settings.

**Please describe your use case for this integration and alternatives you've tried:**

This could be done in the frontend with a custom dashboard, however it would be ideal, if possible, to set these values from esphome instead, similar to how climate components have a `visual:` node.

**Additional context**

OttoWinter, 2021-08-01:
> The number component does not currently have a way to set visual settings, such as a default number, or min/max settings.

But it already does? For example the `template` number exposes exactly that through the `min_value`/`max_value`/`step` properties: https://esphome.io/components/number/template.html

Weissnix4711, 2021-08-01:
> https://esphome.io/components/number/template.html

Sorry, my bad. I did not look at the docs for the template number, I only saw the base number configuration (on the number core page).

closed 2021-08-01 by Weissnix4711