esphome/feature-requests #1328: Add component for number-valued Tuya endpoints that are both readable and writable

alexbarcelo, 2021-08-01:
**Describe the problem you have/What new integration you would like**

I have a ceiling fan driver with a Tuya MCU and I would like to be able to control the number of minutes (datapoint number 7, in my specific product). This number of minutes is published in a certain datapoint (when used through the RF remote). I would like to control the number of remainig minutes from ESPhome/HomeAssistant.

**Please describe your use case for this integration and alternatives you've tried:**

A read-only approach is viable through the use of a tuya sensor. However, the number of minutes should be writable, and the sensor does not allow to change the Tuya endpoint value.

**Additional context**

I believe that, now that Home Assistant as well as ESPhome have a "number" thing, an integration through that same number platform would be the best. A "tuya number" may be a new component that can be used for this and other use cases, I believe.

frankiboy1, 2021-11-17:
I completely agree with the @alexbarcelo. I'm trying to add support for a Tuya siren, which has multiple datapoints to control things like volume, alarm duration, temperature trigger points and so on. Adding support for all devices from Tuya is probably not desired as it would require a lot of effort for both implementation and maintenance. A "tuya number" together with "tuya switch" would probably allow us to support most devices out there.

I'm currently able to control the siren by using multiple tuya switches, tuya fans and tuya dimmers. It kind of work, but it's a mess. Also, I need to do some conversion as the tuya devices are making its own conversions that is not desired. Ex. the dimmer gets the percentage as input and converts this to a value between 0 and 255.

nagyrobi, 2022-07-07:
Implemented as above.