esphome/feature-requests #1338: Riemann Sum intergral

enablingpenguin, 2021-08-02:
**Describe the problem you have/What new integration you would like**

Would it be possible to add a Riemann sum integral component directly into esphome, pretty much doing what home assistant is doing this this integration but directly on the esphome?

**Please describe your use case for this integration and alternatives you've tried:**

Currently calculating watthrs in home assistant

**Additional context**

enablingpenguin, 2021-08-02:
Forgot to link home assistant integration:
https://www.home-assistant.io/integrations/integration/

OttoWinter, 2021-08-02:
See https://esphome.io/components/sensor/integration.html

enablingpenguin, 2021-08-02:
> See https://esphome.io/components/sensor/integration.html

Thank you very much that will work, It does not appear when you search "Riemann" on esphome for any future searchers.

closed 2021-08-02 by enablingpenguin