esphome/feature-requests #1339: "wait_until" with timeout

markormesher, 2021-08-03:
**Describe the problem you have/What new integration you would like**

I would like to extend the `wait_until` automation component with a timeout duration and on-timeout actions. The existing behaviour wouldn't need to change as the timeout and actions should be optional.

For example:

```yaml
script:
  - id: enable_gps
    then:
      - uart.write: "AT+GPS=1\r\n"
      - wait_until:
          condition:
            lambda: return id(gps_data_recd);
          timeout: 10s
          on_timeout:
            then:
              - logger.log: "No reply from GPS within 10 seconds"
              - // error state handling
              - script.stop: enable_gps
      - logger.log: "GPS is ready"
```

In this example the timeout actions don't implicitly prevent the rest of the script from running, but using `script.stop` allows that behaviour to be achieved very easily.

**Please describe your use case for this integration and alternatives you've tried:**

My use case is waiting for responses or state changes when communicating with other devices; if no response is received within some time period it usually indicates an error that should be handled, rather than waiting forever.

This is sort of possible at the moment using delays and multiple scripts, which is fine for one or two instances of this pattern, eg:

```yaml
script:
  - id: enable_gps_timeout
     mode: restart
     then:
      - delay: 10s
      - logger.log: "No reply from GPS within 10 seconds"
      - // error state handling
      - script.stop: enable_gps

  - id: enable_gps
    then:
      - uart.write: "AT+GPS=1\r\n"
      - script.start: enable_gps_timeout
      - wait_until:
          lambda: return id(gps_data_recd);
      - script.stop: enable_gps_timeout
      - logger.log: "GPS is ready"
```

However, some device communication can involve dozens of messages back and forth, which makes things very cumbersome and hard to keep track of.

**Additional context**

It might be problematic that the current `wait_for` block expects the condition as a direct child rather than within a `condition` sub-key like the `if` block does; I'd hope that's something that can be overcome as there are plenty of other blocks that accept different shapes of value.

If this would be a welcome and positive addition to the platform I'd be happy to take a look at implementing it, but would probably need some input from people who know the codebase better than I do.

nagyrobi, 2022-07-08:
Timeout is supported: https://esphome.io/guides/automations.html?highlight=wait_until#wait-until-action

closed 2022-07-08 by nagyrobi

markormesher, 2022-07-08:
Thanks @nagyrobi , I hadn't seen that update. There's still a part of this FR for "on-timeout" actions - I will open a new ticket to track that.