esphome/feature-requests #1344: SolarMax Inverter Integration

gregoinc, 2021-08-06:
**Describe the problem you have/What new integration you would like**
The Domoticz system has functionality to interface with the SolarMax solar inverter via a LAN interface. I now use Home Assistant, however I have no choice but to keep Domoticz to interface with my SolarMax inverter. I want to retire Domoticz, so having the SolarMax integration in HA would enable me to use Home Assistant exclusively.

**Please describe your use case for this integration and alternatives you've tried:**
Domoticz supports the SolarMax solar inverter over the LAN natively, and Home Assistant doesn't have any integration for SolarMax. The only other alternative would be for me to write my own integration code... but I am not that good :-)

**Additional context**
See the reference to SolarMax on the Domoticz page https://www.domoticz.com/wiki/Hardware_Setup

jesserockz, 2021-08-06:
This seems like a feature request for Home Assistant, **not** ESPHome?

gregoinc, 2021-08-07:
> This seems like a feature request for Home Assistant, **not** ESPHome?

Yes, you are absolutely correct. Apology, I will try and find the HA request location.

jesserockz, 2021-08-07:
Looks like you found it =)

Linking for future searches: https://community.home-assistant.io/t/solarmax-inverter-integration/327723

closed 2021-08-07 by jesserockz