esphome/feature-requests #1345: issues with new light model and transition

tungmeister, 2021-08-06:
### The problem

Just updated to hass 2021.8 in order utilise the new lighting model I have some smart bulbs setup as RGBWW lights.
When using ```color_interlock: true``` the hass UI displays as I would expect. No extra sliders for colour and whiteness with the separate tabs for colour and white temperature, selecting a colour will switch to it as expected. The only issue I'm having is that I have a 0.5s transition setup on the light as standard for all their changes. Due to the interlock when switching from WW to RGB there's quite a delay as the light has to turn off then turn on the RGB LEDs which results in a moment of total darkness. 
I appreciate that the interlock's main purpose is for lights that can't output to both RGB and WW channels at one or due to power limitations but without using it it's back to the additional sliders when changing to colour in order for it to display. (which isn't very family/user friendly)
An ideal solution for me would be to have an additional option for interlock such as ```crossfade: true``` which would start the transition to the RGB LEDs at the same time as turning off the WW ones removing the point at which the bulb is completely off. Thoughts?

### Which version of ESPHome has the issue?

1.21 dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp8266

### Component causing the issue

light

oxan, 2021-08-09:
> The previous version of ESPHome I had flashed it with was 1.19.4 and with it, the colour fades in and out as expected, so it seems that this is a bug in the new colour model of 1.20.0.

v1.21 should fix this behaviour.

Tugzrida, 2021-08-09:
Brilliant thank you! :)

vandalon, 2022-06-02:
Since this request is still open I assume the above does not mean it should work? :)

nagyrobi, 2022-07-05:
@vandalon  Did you try?

vandalon, 2022-07-05:
@nagyrobi i'm sorry, I will test this asap and come back to you.

tungmeister, 2022-07-06:
@oxan any progress on this one?