esphome/feature-requests #1347: sim800l: allow to answer incoming calls

andriej, 2021-08-07:
**Describe the problem you have/What new integration you would like**
Add ability to 'catch' numbers calling on SIM800L SIM card

**Please describe your use case for this integration and alternatives you've tried:**
No alternatives available

**Additional context**
It would be nice to be able to i.e. allow family open the gate by just calling SIM800L module with 'authorized' phone number, so template could be used to perform actions after SIM card called from certain number.
It's seen on the market by commercial devices - people (their numbers) are added to allowed lists and after performing a call their call is being dropped and i.e. entrance gateway opens.

I can see recently that @spilin added dial support so it's just a meter away from that being able to be implemented I guess... Hope it's doable

andriej, 2022-06-14:
@glmnet sorry to call you directly, but after seeing call make implemented last things I miss in this component to be part of neighbourhood-gate-opener is call receive
1.
to hang up the call it's I guess just a matter of adding `ATH` command after `RING` received?
Being able to define number of telephones authorized to run automation and hanging up their call just after received would be what commercial units are doing.
They tend not to pick up calls from not authorized numbers too.

2.
about call_make - there's a _flaw_(?), we are not able to define for how long the call should be made.
If someone doesn't hang up, it can be costly ;-) 

Maybe it's worth to add option to define seconds (starting from 0 - hang up after call received)?

andriej, 2022-07-10:
Few more findings that may allow faster implementation:

> Incoming call is usually indicated by ‘RING’ on serial monitor followed by phone number and caller ID. 

> **ATA** – Accepts incoming call.
> **ATH** – Hangs up the call. On hanging up the call it sends NO CARRIER on the serial monitor indicating call couldn’t connect.
[image]
([credits](https://lastminuteengineers.com/sim800l-gsm-module-arduino-tutorial/))

so basically in the loop of status check (where subprocess of SMS received is checked) there's need to check for `RING` then identify and callback the details from CLIP:
[image]
and `ATH` (hang up) the call.
([credits](https://www.sparkfun.com/datasheets/Cellular%20Modules/AT_Commands_Reference_Guide_r0.pdf))

Identification on serial would look like:
`+CLIP: "+<international number here>",145,"",,,0` where there's only need to check for first value (international number) and "0" at the end

glmnet, 2022-07-11:
Thanks, but in my case type is giving `161` instead of `145` so I might just create a trigger with the value of number and let the logic in yaml take it from there

glmnet, 2022-07-11:
https://github.com/esphome/esphome/pull/3630

https://esphome.io/guides/faq.html#how-can-i-test-a-pull-request

closed 2022-09-14 by glmnet