esphome/feature-requests #1352: Delonghi Pinguino Support for the IR Remote Climate component

pedrofracassi, 2021-08-10:
**Describe the problem you have/What new integration you would like**

I just got a Pinguino portable AC and would love to have it controlled through Home Assistant using ESPHome. Problem is it uses a complex protocol where every button press onthe remote sends the whole state to the AC (much like many other ACs)

**Please describe your use case for this integration and alternatives you've tried:**

I tried using the regular IR Transmitter, but it would require me to have very specific states when turning the AC on or changing temperature

**Additional context**

There's an issue on the IRremoteESP8266 project that has the IR protocol for Delonghi ACs reverse engineered already: https://github.com/crankyoldgit/IRremoteESP8266/issues/1096

formatBCE, 2022-08-13:
Case is, that for Penguino (at least mine) remote is "smart", and does send ALL state variables in one command.
I discovered it while setting up Broadlink remote for HA - basically, when you're pushing fan speed button, while AC is in fan only mode, command will be completely different from same button press in A/C mode.

So you will have to remember all different states of remote as separate IR commands.
 It took like an hour for me, generating list like:
- ac_lowfan_25degrees
- ac_lowfan_26degrees
...
- fan_lowfan
- fan_middlefan
- fan_highfan
- fan_autofan
...

And then I just call command from IR smart remote (for ESPHome it will be just IR transmitter).

Yes, a bit inconvenient, but once done, you will be able to change temperature, say, from 30 to 21 in one click.

grob6000, 2022-08-13:
Clever workaround! I ended up creating a new component that does this; has since been merged. See https://github.com/esphome/esphome/pull/3124

closed 2024-06-27 by nagyrobi

tda90, 2024-07-02:
> Broadlink remote for HA

Hello,
I've just bought a Broadlink remote for HA too.
I was able to configure it though "remote:learn_command" , but I have no clue/idea on how to set a temperature.
The Pinguino, is a "smart" one, which has also an APP, from where it's possible.
But the original remote of this one, has only the PLUS/MINUS for temperature.
So - how to read the commands exactly?
Sorry I'm a little new to this kind of things.

Ingo-Klein, 2024-07-03:
@formatBCE please send the code... thx

formatBCE, 2024-07-03:
> @formatBCE please send the code... thx

That setup is long gone. Since then I moved to ESP32 with IR transmitter and ESPHome, getting rid of Broadlink.
Most probably, even the codes themselves wouldn't work for you, you'd have to catch them with custom IR receiver.

Ingo-Klein, 2024-07-03:
okay thx... order@Bookstore done :)