esphome/feature-requests #1354: Pulse Meter debounce rising signal

boarder876, 2021-08-13:
**Describe the problem you have/What new integration you would like**

Adding a debouncing option to the pulse meter sensor just for rising and falling signal seperatly, to avoid a second pulse caused by a bounce effect. The internal_filter can just debounce a complete pulse from rising to next rising. This is just working for short pulses.
[image]

**Please describe your use case for this integration and alternatives you've tried:**
The Pulse_meter function in combination with TCRT5000 sensor on a Ferraris energy counter cause is counting a pulse by getting high and also by getting low. I assume this is caused by bouncing effect of the digital output of the sensor. So every rotation I receive a short pulse (red area) and a long one (time between the red marking). The internal_filter needs to set to 20sec to be able to measure down to 120w, but this will cause that all above 2400W will be ignored.
I've used ESPeasy some time, where the same hardware configuration is working well and counts precise, so my hope is that sensor bouncing could also be solved by software.

**Additional context**

killee, 2021-08-23:
Same issue in my setup!
In my opinion it can be solved. The behavior should be changed.
A rising edge has to be discarded if a falling edge is in the time of the internal_filter time.

Now the last rising edge after the internal_filter time is used. But this is not suitable for the problem above.

boarder876, 2021-08-24:
ESPEasy is using different modes to select, edge and pulse mode. Especially the pulse mode is working well as expected, but I like esphome more than espeasy.
Here's teh description from the documentation about the difference and their application:

For **Edge** mode types, the counter is incremented when a rising, falling signal edge or either of these (change) is detected at the selected GPIO (GPIO interrupts are used). BUT incrementing is only done, if the time between most recent counting and current edge is longer than the selected Debounce Time).

These **Edge** mode types are best suited for high frequency signals with quite constant frequency so that debounce time can be adjusted to that. Counting of 40 cycles/second (2400 rpm with Debounce time =25ms) should be possible or possibly up to 4 times faster (Debounce time = 0), depending on the cpu workload through other ESPEasy tasks and WiFi handling. However the electrical signal must (when Debounce Time had ended) be stable (well filtered and no crossover of signals or signal spikes). Otherwise there is a tendency of counting slightly too much pulses.

For **Pulse** mode types the pulse detection is also initiated by the rising, falling or change signal edge, but it is only counted as a pulse, if the GPIO signal is stable over three consecutive GPIO level samples. First, after an edge was detected, we wait for the Debounce Time, and then the GPIO level is checked three times with two intermediate delays of half the debounce time. Only when all three times the level is identical the signal is counted as a pulse. So the minimum pulse length is Debounce Time x 2.. Otherwise the three checks are repeated until a sable signal is detected.

The **Pulse** mode types are best suited for lower frequency signals (>80ms=12,5 cycles/second=750prm with Debounce Time >20ms) and also can count reliably very low frequencies and long pulses of minutes and hours (e.g. an gas meter that does not send any signal, when all consumers are off for a long period). Electrical interferences (shorter than Debounce Time) are well filtered out.

nagyrobi, 2022-07-06:
Implemented with `internal_filter_mode` configuration, see https://esphome.io/components/sensor/pulse_meter.html