esphome/feature-requests #1356: Add support for Energy management and long term statistic in Teleinfo device

DrCoolzic, 2021-08-15:
**Describe the problem you have/What new integration you would like**

The 2021.8.0 release add very nice features for **Energy management** and **long term statistic** but unfortunately it does not seems to be possible to use directly any of these features with the sensors created with the teleinfo device. 

I guess that the sensors created by the **teleinfo** integration are missing specific properties to use them directly in energy management and long term statistics.

**Please describe your use case for this integration and alternatives you've tried:**

For energy measurement I have defined this in the configuration:
```
  - platform: teleinfo
    tag_name: "EAST"
    name: "linky_east"
    unit_of_measurement: "Wh"
    device_class: "energy"
    state_class: "measurement"
    icon: mdi:flash
```
But unfortunately it cannot be used directly as an energy source (probably missing the last_reset attribute?).

The workaround I found is to use the following integration
```
utility_meter:
  energy_day:
    source: sensor.linky_east
    cycle: daily
```
as a source for the energy and it works, but nevertheless it is only a workaround!
___

As mentioned the second nice feature is long term statistics. So I have tried to define a sensor like this:
```
  - platform: teleinfo
    tag_name: "IRMS1"
    name: "linky_irms1"
    device_class: "current"
    state_class: "measurement"
    unit_of_measurement: "A"
    icon: mdi:flash

```
The sensor seems to contain all the requested information for long term statistics (see https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics) but it does not work (cannot be added to lovelace ui).
Seems like some extra information is needed?

**Additional context**

DrCoolzic, 2021-08-23:
It seems like last_reset attribute will be dropped in 2021.9 
see https://github.com/home-assistant/core/pull/54523
However the new mechanism for energy management will be based on 2 new state_class type.
Therefore I have run a test with a sensor like this
```
sensor:
  - platform: teleinfo
    tag_name: "EASF01"
    name: "linky_easf01"
    unit_of_measurement: "Wh"
    device_class: "energy"
    state_class: "total_increasing"
    icon: mdi:flash
```
This sensor compiles correctly however if I look at the sensor in the developer tools the _state_class: "total_increasing"_
is not present

[image]

**I guess esphome needs to add support for this new class**

manu3b1, 2021-09-04:
I found a way to make it work, using **last_reset_type: auto** forces **state_class** to **total_increasing** on HS side and generates **last_reset** info.
```
sensor:
  - platform: teleinfo
    tag_name: "HCHC"
    name: "Linky HC"
    unit_of_measurement: kWh
    device_class: energy
    state_class: measurement
    last_reset_type: auto
    filters:
      - multiply: 0.001
    accuracy_decimals: 3
    icon: mdi:flash
    teleinfo_id: linky
```

Result entity:
```
state_class: measurement
last_reset: 2021-09-04T09:28:18.793609+00:00
unit_of_measurement: kWh
friendly_name: Linky HC
device_class: energy
```

DrCoolzic, 2021-09-04:
Thanks for the information quite interesting.

However, we do not need this anymore. The 2021.9 release solve all the problems. Now for energy you need to specify a state_class “total_increasing” and in that case you do not need any more of the last_reset field.

So to use the total active energy drawn off as the input for “grid consumption” you need this kind of configuration (in standard mode):
```
Sensor:
  - platform: teleinfo
    tag_name: "EAST"
    name: "linky_east"
    unit_of_measurement: "Wh"
    device_class: "energy"
    state_class: "total_increasing"
    icon: mdi:flash
```
This gives you:
```
state_class: total_increasing
unit_of_measurement: Wh
friendly_name: Total energy
device_class: energy
```
And if you have solar panel you can input the energy retuned to the grid with a configuration like this
```
  - platform: teleinfo
    tag_name: "EAIT"
    name: "linky_eait"
    device_class: "energy"
    state_class: "total_increasing"
    unit_of_measurement: "Wh"
    icon: mdi:flash
```

For the usage of long term statistic type of sensors you also need to specify the sensor using the proper device_class and state_class. For example for Voltage and Current
```
  - platform: teleinfo
    tag_name: "IRMS1"
    name: "linky_irms1"
    device_class: "current"
    state_class: "measurement"
    unit_of_measurement: "A"
 - platform: teleinfo
    tag_name: "URMS1"
    name: "linky_urms1"
    device_class: "voltage"
    state_class: "measurement"
    unit_of_measurement: "V"
```

By using this type of configuration, you can display the long time statistic using a statistic card
[image]
[image]

Currently the graph shows only two days of statistical data as the feature has just been release :)

Therefore it seems that everything is available in the current release of teleinfo, but nowhere it has been properly documented. May be it is fine to let the user do trials and errors to find the correct way to specify the different type of sensors or may be **this needs to be included in the documentation of next teleinfo component release.**

But It does not seems that there is much activity for teleinfo?

DrCoolzic, 2021-09-04:
May this request should be closed and a new one opened for documentation?

closed 2022-07-01 by nagyrobi