esphome/feature-requests #1360: name_add_mac_suffix feature to extend to wifi.ap.ssid name also

meijerwynand, 2021-08-16:
**Describe the problem you have/What new integration you would like**

A similar option to `esphome.name_add_mac_suffix` [option](https://esphome.io/components/esphome.html#esphome-mac-suffix) must be available in the [WiFi component](https://esphome.io/components/wifi.html) and `wifi.ap.ssid` to create a dynamic SSID.

**Please describe your use case for this integration and alternatives you've tried:**

In the event several LED WiFi light bulbs need to be configured, and their config contains an AP fallback, they can easily be distinguished as separate devices instead of repeated identical names when scanning for WiFi devices.

* A `wifi.add_mac_suffix` option is proposed to add the suffix to the given SSID. Thus an SSID of `lightbulb` can be detected in the WiFi network as `lightbulb-a12bbc`. Here the MAC suffix as added to the AP SSID.
* This would allow all devices to be flashed with the same `lightbulb.yaml` instead of multiple seperate `yaml` files.
* It can also serve as a easier method to just connect to the device (then in hotspot mode) and upload a relevant bin file. 

```
substitutions:
  name: lightbulb

esphome:
  name: ${name}
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout
  name_add_mac_suffix: true

wifi:
  ap:
    ssid: ap_${name}
    password: Abcd4321
    ap_timeout: 1min
    add_mac_suffix: true

captive_portal:
```
Looking forward to your thoughts on this suggestion.

Thanks

**Additional context**

h0jeZvgoxFepBQ2C, 2022-09-19:
did you find a solution for this?

tiredofit, 2024-06-13:
This would be a very useful and beneficial addition in environments with devices that follow similar templates to determine which device you are actually connecting to. Yes, one could read the BSSID via a scanner or alternate means, however this would add significant time to configure this manual connection to the same AP SSID with a specific BSSID especially if you are trying to race towards a 1 minute timeout.

butuletz, 2024-07-05:
I was looking for a solution to this issue as well, and I couldn't find a similar one similar to `esphome.name_add_mac_suffix` to easily identify the AP based on the mac of the device. 

However, I looked at the **esphome** repository and the actual code.... 

It turns out that if you omit the `wifi.ap.name` option in the YAML configuration, and you have `esphome.name_add_mac_suffix` option set to `true`, the AP name will be the actual device name (which includes mac address), so exactly what we are looking for! 😃 

Hope this helps someone in the same situation. Cheers!

Arafa-microsys, 2025-02-27:
> I was looking for a solution to this issue as well, and I couldn't find a similar one to `esphome.name_add_mac_suffix` so I can easily identify the AP SSID for a particular device based on the mac of the device.
> 
> However, I looked at the **esphome** repository and the actual code....
> 
> It turns out that if you omit the `wifi.ap.name` option in the YAML configuration, and you have `esphome.name_add_mac_suffix` option set to `true`, the AP name will be the actual device name (which includes mac address), so exactly what we are looking for! 😃
> 
> Hope this helps someone in the same situation. Cheers!

Thank you, this is the solution, when defining, you must not enter SSID in AP
just

```
ap:
  id: my_wifi  # Assign an ID to reference the WiFi component
  power_save_mode: none
captive_portal:
```