esphome/feature-requests #1362: Publish power consume of the device if it is on, to be used on Home Assistant

morenod, 2021-08-18:
**Describe the problem you have/What new integration you would like**

Publish power consume of the device (manually entered) when it is on, to be added on new home assistant power manager

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nuttytree, 2021-08-19:
@morenod I have actually implemented this in a couple of custom components that I use, take a look at my [README](https://github.com/nuttytree/ESPHome-Devices#custom-components).  I thing to note is that this approach does have some limitations, for example for dimmers the power usage varies based on the brightness level but the ratio of brightness to power is likely different from dimmer to dimmer and very possible a non-linear function.  Even for straight on/off devices it can be problematic, for example I thought about using this for my coffee make but the heating element is not on the entire time the coffee maker is on so the power usage is not consistent.

rradar, 2021-08-21:
Actually this can be already achieved with esphome right away by just using a template sensor:

```yaml
switch:
  - platform: output
    name: "Fan"
    output: relay
    on_turn_on:
    - sensor.template.publish:
        id: fan_assumed_power
        state: 10
    on_turn_off:
    - sensor.template.publish:
        id: fan_assumed_power
        state: 0

sensors:
  - platform: template
    name: "Sala Fan assumed Power"
    id: fan_assumed_power
    unit_of_measurement: "W"
```

nuttytree, 2021-08-21:
The big disadvantage of the template sensor is that it only publishes a value when it changes so if you then use the Total Daily Energy or Integration sensor to create an energy (Wh) from it the total doesn't get updated until you turn off the device.

nuttytree, 2021-08-30:
FYI, I just put up a PR to resolve the issue with using a template sensor for this purpose esphome/esphome#2224.

rradar, 2021-09-04:
For now I just hacked the missing publishing with a little script:
```yaml
script:
  - id: fan_assumed_power_publisher
    then:
    - while:
        condition:
          switch.is_on: switch_fan
        then:
        - sensor.template.publish:
            id: an_assumed_power
            state: "10"
        - delay: 10s
```
thank's to your pr @nuttytree I probably don't need this anymore in the next esphome release :rocket:

closed 2022-07-01 by nagyrobi