esphome/feature-requests #1363: "Update all" updates only online devices

TheGroundZero, 2021-08-18:
**Describe the problem you have/What new integration you would like**

The `Update All` feature of ESPHome allows for quick updating of all linked devices.
However, this also results in the building of firmware images and the needless waiting for network connection with devices that are currently offline.
This creates unnecessary delays.
Some devices may even be offline for multiple version upgrades because they're only used for sporadic testing.

**Please describe your use case for this integration and alternatives you've tried:**

There's currently 2 options:
- Manually update each of the online devices
- Update All but which takes extra long

**Additional context**

Request #1317, which asks for an option to update only selected devices could be a solution to this as well.
Manually select the online devices and update those.

ipetrovits, 2021-08-18:
I don't want to hijack your idea, but in some use cases a "Compile all/selected" option would be useful as well.. I'm compiling all of my devices right now on my pc via CLI, then updating only the online ones one-by-one from the raspberry.

sbur83, 2023-01-27:
I had the same request