esphome/feature-requests #1364: Add support for TDS and water flow sensors (like SEN0244)

DorjeDorf, 2021-08-18:
**Describe the problem you have/What new integration you would like**

DFRobot have a pretty nifty TDS sensor that can be used together with flow meter like this one. 
It would be awesome if ESPHome would give support for such devices.

**Please describe your use case for this integration and alternatives you've tried:**

These sensors can be great to detect when a Reverse Osmosis membrane is used up and safe guards agains too much water waste. It can also alert you to when you should exchange generic filters (based on rising TDS numbers). 
It can be done now via Arduino and other platforms like MQTT or MySensors, but I'd love to have it all nice and tidy via ESPHome. 

**Additional context**

DFRobot product info [here](https://wiki.dfrobot.com/Gravity__Analog_TDS_Sensor___Meter_For_Arduino_SKU__SEN0244)
Sample flow meter/TDS sensor [here](https://www.aliexpress.com/item/32870871997.html)

realtek2017, 2021-09-25:
This would be great to have something natively supported although because it works using the analogue signal it can already be used in ESPhome. You just need to add a lamdba filter to do the conversion calculation. 

I have been testing this and I am getting some incorrect readings using the standard calculation that these TDS meters use. I will report back once I have a working example

fokcuk, 2022-02-27:
Did you get anywhere with this @realtek2017 ?
Starting to look at it just now

theNorr, 2024-05-14:
I'd like this too. I saw an article on instructables.com where a guy did a water quality monitoring device using DFRobot sensors for TDS, PH and ORP together with an Arduino that was running a ML model to analyze and determine the quality. The result where then transmitted over/with GSM and SMS. It would probably be much simpler using together with HA and ESPHome, and would open up a lot of possibilities.

See https://www.instructables.com/GSM-SMS-Enabled-AI-driven-TinyML-Water-Pollution-M/