esphome/feature-requests #1365: Ir remote climate control missing Midea IR

TitooNL1981, 2021-08-18:
**Describe the problem you have/What new integration you would like**
I'm missing Ir remote climate controle midea Ir, it there as uart, but nog as IR. mine Midea has no Uart.

**Please describe your use case for this integration and alternatives you've tried:**
Missing Midea as Ir climate control. It is available as espeasy and Arduino 

https://github.com/ToniA/arduino-heatpumpir
https://espeasy.readthedocs.io/en/latest/Plugin/P088.html#p088-page

i'm using now Espeasy but the Esphome homeassistant looks cool!

**Additional context**

jesserockz, 2021-08-18:
You can try out support from this PR: https://github.com/esphome/esphome/pull/1343

bwze, 2021-10-04:
Would love to have this too. 

I see the new midea ac component has an option for transmitter_id which is essentially what we're looking for, but I can't get it to work properly and I wonder if it's because this component was created specifically for the Wi-Fi to UART dongle solution. 

In my case, I have an older Midea that doesn't have any option for connection to Wi-Fi....it simply used IR for control and I know there happens to be compatibility for this brand with the IRremoteESP8266 library.

nagyrobi, 2022-07-06:
Implemented in https://esphome.io/components/climate/climate_ir.html#midea-ir-climate

closed 2022-07-06 by nagyrobi