esphome/feature-requests #1366: Allow unit_of_measurement on template number

JamesSwift, 2021-08-19:
**Describe the problem you have/What new integration you would like**

Several of my devices have variables/readings that I need to be able to configure from home assistant, as well as update from the esp internally. For example: the amount of food a cat is allowed, the weight of a cat.

The new `Template Number` type if ideal for this. The only thing lacking is the ability to specify a unit of measurement. Home assistant's `input_number` allows you to specify a unit of measurement, it would be really nice to be able to display that for the `number` type as well. I presume this would require changes from ESPHome and home assistant to support.

**Please describe your use case for this integration and alternatives you've tried:**

When displaying `number`s next to `input_number`s or `sensor`s, the unit of measurement is obviously missing.

**Additional context**

In the past I used a `global` variable, a `template sensor`, and a custom `api service` on the esp, along with a matching `input_number`, and `automation` on home assistant to allow the value to be changed and synced from either end. This also allowed a unit of measurement to be specified. I am in the process of transitioning to the new `template number` type as it is much cleaner.

[image]

Tuckie, 2021-10-23:
Also interested in this!

nagyrobi, 2021-12-31:
This works in the current version.

Tuckie, 2022-01-12:
> This works in the current version.

could you show me where?: https://esphome.io/components/number/template.html

nagyrobi, 2022-01-12:
I use this:
```
number:
- platform: template
  icon: "mdi:fan-speed-1"
  name: "Alap elszívás/befúvás"
  entity_category: config
  optimistic: true
  restore_value: true
  initial_value: 30
  unit_of_measurement: "%"
  id: lg_300_speed_1
  min_value: 10
  max_value: 45
  step: 1
```
Resulting:
[image]

MagMeister, 2022-05-23:
I confirm that unit_of_measurement is shown in the HomeAssistant interface, but not in the ESPHome web server.
Is it possible to add it there as well?

nagyrobi, 2022-07-01:
OP's request fulfilled

closed 2022-07-01 by nagyrobi