esphome/feature-requests #1373: Optional logs viewing timeout for the "run" command

nmaggioni, 2021-08-24:
**Describe the problem you have/What new integration you would like**

When using the `run` command of the CLI utility, logs are shown indefinitely and sending a signal (e.g.: CTRL+C'ing) is necessary to quit the process. An optional flag to specify a maximum time to show logs for before terminating automatically would be a nice addition.

**Please describe your use case for this integration and alternatives you've tried:**

When upgrading my fleet of ESPHome devices I use a simple script that cycles through my YAML configs and updates each unit through the `run` command. This makes it necessary to manually terminate each log viewing session to proceed to the next module.

An optional timeout that only starts when the module is reachable and logs starts streaming would be handy to automate this kind of upgrade and monitoring processes.

Using a sequence of `compile` and `upload` commands is already possible, but it doesn't give any information on the outcome of the upgrade and the status of the module after the reboot.

**Additional context**

Excerpt of the update script:
```bash
for conf in *.yaml; do
    if [ -d "${conf%.yaml}" ]; then
        docker run --rm -v "${PWD}":/config -it "esphome/esphome:$ESPHOME_VERSION" "$conf" run
    fi
done
```

nmaggioni, 2021-08-28:
Your approach will terminate after 15s whether the module came back up after the flash or not. If something went wrong, connection will be attempted for some time and then the script will happily proceed with the next config anyway.

WeekendWarrior1, 2021-08-28:
> Your approach will terminate after 15s whether the module came back up after the flash or not.

> An optional timeout that only starts when the module is reachable and logs starts streaming would be handy to automate this kind of upgrade and monitoring processes.

Is what you actually want a nice way to confirm that a **module is reachable**, and the timeout option to run is just one way of doing that? What do you want to happen if a module doesn't become reachable?

> With a timeout (proposed behavior):
> 
>     For each config file:
>         esphome config.yaml run --timeout=5
>         Module gets flashed OTA and reboots.
>         The CLI waits until the module comes back online.
>         Logs start streaming.
>         After 5s the process terminates.
>         Loop over to the next config file.

Is this the functionality you actually mean, and the logs are just a convenient way to make sure a device is back online?:

- For each config file:
  - esphome config.yaml run 
  - Module gets flashed OTA and reboots.
    - You want to a way to nicely check if the device has come back online
      - if device not come back online, stop here
      - else 
        - Loop over to the next config file.

nmaggioni, 2021-08-30:
> Is this the functionality you actually mean, and the logs are just a convenient way to make sure a device is back online?

More or less, yes. A way of telling that the module got flashed correctly, rebooted, has connected to the network, and the new config/version isn't creating obvious new problems (components spitting errors, misconfigurations that can't be detected at compile time...).

A more correct indicator would be, for example, the reset of the boot loop counter - even if extracting that information from the CLI sounds way more difficult and/or error prone than the aforementioned approach.