esphome/feature-requests #1375: Add support for Alpicool Fridge

crozo5, 2021-08-24:
**Describe the problem you have/What new integration you would like**
I wanna that ESPHome support [Alpicool](https://www.alpicool.com/index.html/) fridges.

**Please describe your use case for this integration and alternatives you've tried:**
Alpicool refrigerators are the most used for people with camper vans or caravans due to their quality / price ratio. These refrigerators have Bluetooth to be controlled through an [APP](https://play.google.com/store/apps/details?id=com.alpicoolneutral.fridge.controller&gl=ES). It would be great to be able to control the on/off and temperature control from anywhere.

**Additional context**

johnelliott, 2021-09-27:
I've done some research, and my software may work with home assistant without any modifications. There's a part of home assistant to interact with apple homekit devices that may work. I don't use HA, but if someone else with an Alpicool fridge wants to help, perhaps we can get this working.

clomads, 2022-06-13:
I’d love to help with this, I have an older Alpicool w/out bluetooth, but am about to upgrade to a larger one with. I have several ESP32 devices running in my bus with ESPHome that I can test on.

sixtycyclehum, 2022-07-21:
I just bought "EUHOMY" which as far as I can tell is an Alpicool and has the same app.  If we could get an ESP32 library going, seems ripe for an alpicool bluetooth gateway so folks can integrate it on things like Blynk or node JS.  Certainly small enough to integrate it right into the bowels of the fridge.

johnelliott, 2022-07-21:
I'm getting another Bluetooth Alpicool this week (mine was stolen 😤). I recently picked up an esp32 board, so perhaps there will be be a critical mass of interest now.

sixtycyclehum, 2022-07-21:
> 

Awesome (except for the theft - that sucks)!  I had my eye on your other project with HomeKit, but wasn't sure if I had enough expertise to dive in.  I'm happy to help if I can keep up.

markgdev, 2022-07-30:
I'm interested in helping out with this. Not done any esphome dev before but happy to try and pick it up as I go along.

black-jmyntrn, 2022-08-04:
I am soooo game for this! I have home assistant running in my Overland Prius and an Alpicool fridge. 
[MY PRIUS HAS A REFRIDGERATOR](https://black.jmyntrn.com/2020/12/30/my-prius-has-a-refridgerator/)
[THE BASE OF MY CAR AUTOMATION SYSTEM](https://black.jmyntrn.com/2021/10/26/the-base-of-my-car-automation-system/)

markgdev, 2022-08-05:
Does anyone have enough esphome experience to get us started?

johnelliott, 2022-08-05:
I don't have any experience with esp32 or hass. I'm reading some basic docs to get my head around what esphome is, what it does, how it works.

crozo5, 2022-08-18:
Hi!
Any progress on this topic?

black-jmyntrn, 2022-10-01:
> I've done some research, and my software may work with home assistants without any modifications. There's a part of home assistant to interact with apple home kit devices that may work. I don't use HA, but if someone else with an Alpicool fridge wants to help, perhaps we can get this working.

what do you need to get started? I have unlimited resources and am happy to help. 

thing is now... based on what I've gathered you might be our knight in shining armor to get this working for us. just, the way my brain works, if you at least get on base, for most of us we'd consider it a home run and my brain has two other "systems" that with you mirroring success would be very beneficial for A LARGE number of folks out there. 

the current setup in my rig there is a raspi 4(storage nas for better words), raspi 3b+(openwrt main router w/load balancing multiple "hotspot" devices and main Wifi6 Wifi master), beryl router, slate router and a bmax mini pc(zwave usb controller and my home assistant controller). the thing is, my fridge and a few other components that are smart with bluetooth are lacking connectivity to home assistant. this is where the way my brian works is going... if you contact me through my site[black.jmyntrn.com], id love to chat offline and see if we can at least get this alpicool project finished in the next 14 days.

wwj1978, 2022-12-16:
+1 I was trying to reverse engineer my travel cooler which is Yeticool brand (Polish) and I think It's generally the re-branded Alpicool. I'd love to be able to integrate It via MQTT.

kozlik07, 2023-06-20:
hello just bought apicool clone exactly peme expedition uses the same app i also have homeassistan and esp32 i can help but i don't know how to go about it

appi1, 2023-06-20:
hy, i have a Outwell Arctic Cooler with the same Coller-Controller and BT. I'am als very interessted to help in this Project.

markgdev, 2024-05-25:
> The bluetooth implementation changed quite a bit in all these years. Is someone working on this?

I have some python that works outside of esphome/ha, will try and tidy it up and get it into a home assistant integration sometime soon

c0ff3e, 2024-06-17:
> > The bluetooth implementation changed quite a bit in all these years. Is someone working on this?
> 
> I have some python that works outside of esphome/ha, will try and tidy it up and get it into a home assistant integration sometime soon

Let me know how I can help. 

Have all needed items ready for testing.

c0ff3e, 2024-06-24:
> > The bluetooth implementation changed quite a bit in all these years. Is someone working on this?
> 
> I have some python that works outside of esphome/ha, will try and tidy it up and get it into a home assistant integration sometime soon
@markgdev 
Happy to help with the pyton if you care to share.

spacesuitdiver, 2024-06-27:
I got my GPTs fired up for this 😅 I have a BougeRV that I bet uses the same app I was going to see if I could reverse engineer for funs.

gtporter, 2024-06-28:
I have a Renogy cooler that I could use to test.  I think they all use the same control board, BT protocol and app.

jakub-hajek, 2024-11-11:
Maybe somebody finds useful my implementation based on esp32 and mqqt. I do not use esphome yet, but feel free to reuse fridge class. Control commands are taken from [johnelliott](https://github.com/johnelliott) Go implementation.

Gruni22, 2025-07-14:
Maybe also useful for others, my implementation as HA custom component.
https://github.com/Gruni22/alpicool_ha_ble

markgdev, 2025-07-14:
> Maybe also useful for others, my implementation as HA custom component.
> https://github.com/Gruni22/alpicool_ha_ble
> 

This looks great, looking forward to giving it a try

gtporter, 2025-07-14:
You guys are awesome!  I'll try it out with my Renogy dual-zone.

c0ff3e, 2025-07-14:
> There ist still a Bug with negative Temps. I will fix it tomorrow.

I am testing this out, awesome work.
Seems to have a issue with connecting.  Will create a issue if i can't
solve it.

Br
Christoffer

scadman1, 2025-08-20:
I am trying the HACS integration but can't seem to connect to the fridge. How do I find the "address" the app requires?