esphome/feature-requests #1377: Add support for cycling through fan speeds via a service

dgtal1, 2021-08-25:
**Describe the problem you have/What new integration you would like**

I'd like to have a possibility to locally change fan speed via a microswitch in my ESP32 board. Currently I think this isn't possible as the only services for fan are: `toggle`, `turn_on`, `turn_off`.

The new service would be something like: `fan.cycle_speed`. So assuming I'll have 4 speed levels (speed_count) then starting with speed = 1 (25%) the subsequent service calls would set the following speeds: 50%, 75%, 100%, 0%, 25%, 50% etc.

**Please describe your use case for this integration and alternatives you've tried:**

I'm not sure, perhaps there's a workaround by calling `output.set_level`??

**Additional context**

nuttytree, 2021-09-01:
It does require a lambda but you can easily add your own service to do this:

```yaml
api:
  services:
    - service: cycle_speed
      then:
        - lambda: >-
            int speed = id(my_fan).speed + 1;
            if (speed > 4) {
              auto call = id(my_fan).turn_off();
              call.perform();
            } else {
              auto call = id(my_fan).turn_on();
              call.set_speed(speed);
              call.perform();
            }
```

See the [API documentation](https://esphome.io/components/api.html#user-defined-services).

nuttytree, 2021-09-01:
If you wanted to get really fancy you could even read the configured number of speeds rather than hard coding it:
```yaml
api:
  services:
    - service: cycle_speed
      then:
        - lambda: >-
            int speed = id(my_fan).speed + 1;
            int max_speed = id(my_fan).get_traits().supported_speed_count();
            if (speed > max_speed) {
              auto call = id(my_fan).turn_off();
              call.perform();
            } else {
              auto call = id(my_fan).turn_on();
              call.set_speed(speed);
              call.perform();
            }
```

dgtal1, 2021-09-01:
oh, that looks really nice, thanks! 
I'd still vote for having this as a built in functionality in ESPhome, but if it's possible this way with custom services - then it's also fine. 
One question though: can I call these custom services locally, and not from Home Assistant? I mean something like this:
```
binary_sensor:
  - platform: gpio  
    # ...
    on_press:
    - cycle_speed: my_fan
```
Is something like this possible? What does the syntax should look like to call such local service? It's nice that it's being exposed to HA, but I'd like this to work locally even if the node is not connected to HA.

nuttytree, 2021-09-01:
You could put the logic in a script then you could call the script from the service definition that is exposed to HA or from another automation in ESPHome (button click, time, etc.)

nuttytree, 2021-09-01:
I think making this a HA service is probably not inline with the general architecture of ESPHome but it wouldn't take much to make this an ESPHome action.  That would make it available in automations and lambdas and could then be easily added as an HA service if the end user desires.  @jesserockz do you have any opinions on making this an Action in ESPHome?

jesserockz, 2021-09-01:
An action would be fine to create

closed 2021-09-26 by jesserockz