esphome/feature-requests #1386: HLW8032 power sensor implementation request

turbatus, 2021-08-30:
**Describe the problem you have/What new integration you would like**

I installed esphome on my Gosund P1 power strip (8266/8265 controller) and the power meter is not working, showing always 0.0 values. After some research I discovered that it has a different power sensor than the usual model. Instead of a CSE7766 it has HLW8032 (still uses the serial communication, though). I would like to request to add HLW8032 to esphome, if possible.
**Please describe your use case for this integration and alternatives you've tried:**

As mentioned before, the HLW8032 power metering sensor is not yet implemented in esphome. This is the sensor used by the newer Gosund P1 power strips. All the returned values are 0 (using CSE7766 component).
**Additional context**

I already tried with Tasmota, same behavior, as it uses the CSE7766 configuration.
Example Github repo on HLW8032:
https://github.com/Khaalidi/HLW8032
If you can guide me on how to extract relevant info I would be more than happy to do it.
Thank you for your support and effort!
Cheers!

nagyrobi, 2022-07-08:
did you try https://esphome.io/components/sensor/hlw8012.html ?

turbatus, 2022-08-22:
> did you try https://esphome.io/components/sensor/hlw8012.html ?

Yes, I have tried that, did not work. Returns 0.0 value all the time.

gsaiz, 2022-09-08:
I second this request, the new model for the Athom smart plugs [Tasmota EU Plug V2](https://www.athom.tech/blank-1/EU-plug) also comes with the HLW8032 power sensor. It comes pre-flashed with Tasmota.

Tasmota seems to have support [for the HLW8032](https://github.com/arendst/Tasmota/blob/e6b4a77abd6621dd49fac12ccfa88cc7f1742983/tasmota/tasmota_xnrg_energy/xnrg_02_cse7766.ino) [since 2019](HLW8032).

gsaiz, 2022-09-09:
According to Athom’s support, this device is working using the cse7766 as shown in this example:

https://github.com/athom-tech/athom-configs/blob/main/athom-smart-plug-v2.yaml

pencilhead1, 2023-06-13:
I have a Deta GPO that has this chip.
The chip only require RX.  Ensure correct RX pin is config.
This code works for me.

```
#HLW8032
  - platform: cse7766
    update_interval: 15s
    current:
      name: "${friendly_name} Current"
      icon: mdi:current-ac
      filters:
        - multiply: 4.869
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10
            send_first_at: 1
    voltage:
      name: "${friendly_name} Voltage"
      icon: mdi:flash
      filters:
        - multiply: 1.925

    power:
      name: "${friendly_name} Power"
      icon: mdi:gauge
      id: ${friendly_name}_wattage
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10
            send_first_at: 1

    energy:
      name: "${friendly_name} energy"
      icon: mdi:gauge
      id: ${friendly_name}_energy
      filters:
        - sliding_window_moving_average:
            window_size: 10
            send_every: 10
            send_first_at: 1

  - platform: total_daily_energy
    name: "${friendly_name} Daily Energy"
    power_id: ${friendly_name}_wattage
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    icon: mdi:chart-bar
```

phoenixxie0, 2025-02-24:
> voltage:
>       name: "${friendly_name} Voltage"
>       icon: mdi:flash
>       filters:
>         - multiply: 1.925

> I have a Deta GPO that has this chip. The chip only require RX. Ensure correct RX pin in config. This code works for me. I have WB2S chipset.
> 
> ```
> uart:
>   rx_pin: P10
>   baud_rate: 4800
>   
> #HLW8032
>   - platform: cse7766
>     update_interval: 15s
>     current:
>       name: "${friendly_name} Current"
>       icon: mdi:current-ac
>       filters:
>         - multiply: 4.869
>         - sliding_window_moving_average:
>             window_size: 10
>             send_every: 10
>             send_first_at: 1
>     voltage:
>       name: "${friendly_name} Voltage"
>       icon: mdi:flash
>       filters:
>         - multiply: 1.925
> 
>     power:
>       name: "${friendly_name} Power"
>       icon: mdi:gauge
>       id: ${friendly_name}_wattage
>       filters:
>         - sliding_window_moving_average:
>             window_size: 10
>             send_every: 10
>             send_first_at: 1
> 
>     energy:
>       name: "${friendly_name} energy"
>       icon: mdi:gauge
>       id: ${friendly_name}_energy
>       filters:
>         - sliding_window_moving_average:
>             window_size: 10
>             send_every: 10
>             send_first_at: 1
> 
>   - platform: total_daily_energy
>     name: "${friendly_name} Daily Energy"
>     power_id: ${friendly_name}_wattage
>     filters:
>       - multiply: 0.001
>     unit_of_measurement: kWh
>     icon: mdi:chart-bar
> ```

Why  the voltage has  filters:
>         - multiply: 1.925

pencilhead1, 2025-02-24:
I was getting incorrect voltage. I put a multi-meter across inpur terminals.
Trial and error -  1.925 was close enough to actual voltage for my device.

phoenixxie0, 2025-02-24:
> I was getting incorrect voltage. I put a multi-meter across inpur terminals. Trial and error - 1.925 was close enough to actual voltage for my device.

There are 470K*4=1880K and 1000Ω .  let me set it to 1.88