esphome/feature-requests #1389: AC Voltage with ESP32 and ZMPT101B

Pro-Control, 2021-08-31:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

 I try to use  ESP32 and ZMPT101B with EmonLib on Arduino IDE it works 
 but after convert it to Custom Sensor under ESPHome get unstable reding 
Power.h file 
`#include "esphome.h"
#include "EmonLib.h"

class Volt : public PollingComponent, public Sensor {
	
 public:
  
  EnergyMonitor main;
  EnergyMonitor g_one;
  EnergyMonitor g_two;
  Sensor *main_v = new Sensor();
  Sensor *g1_v = new Sensor();
  Sensor *g2_v = new Sensor();
  Volt() : PollingComponent(1500) {}

  void setup() override {
   main.voltage(36, 45, 1.7);
   g_one.voltage(39, 40, 1.7);
   g_two.voltage(34, 35, 1.7);  
  }

  void update() override {
	  
   main.calcVI(20,2000);
   float main_f  = main.Vrms;
   main_v->publish_state(main_f);
   
   
   g_one.calcVI(20,2000);
   float g_one_f  = g_one.Vrms;
   g1_v->publish_state(g_one_f);
   
   
   g_two.calcVI(20,2000);
   float g_two_f  = g_two.Vrms;
   g2_v->publish_state(g_two_f);
   
  }
  
power.yaml
 }; `

`sensor:
- platform: custom
  lambda: |-
    auto voltage = new Volt();
    App.register_component(voltage);
    return {voltage->main_v, voltage->g1_v, voltage->g2_v};

  sensors:
   - name: "Main Volt"
     id: main_volt
     unit_of_measurement: V
     accuracy_decimals: 2
    
   - name: "G1 Volt"
     id: g1_volt
     unit_of_measurement: V
     accuracy_decimals: 2
    
   - name: "G2 Volt"
     id: g2_volt
     unit_of_measurement: V
     accuracy_decimals: 2  `

**Additional context**

Stybyk, 2023-02-25:
This one would be nice to have officially integrated.

Stybyk, 2023-02-26:
I found developed a solution here: [https://github.com/danpeig/ESP32EnergyMonitor](url) , would be good to add this into supported integration.

alwwwex, 2024-02-09:
That's will be nice to have! Trying to get it work via custom component, and it's not simple. Almost everyone using just CT clamp for current and hardcoded value of grid voltage to calculate Watt's. CT current clamps working fine, but to integrate ZMPT101B - no solution "from box".

gaitolini, 2024-03-12:
Interessante ler os comentários, bastante informação....
Estou tentando  criar um componente externo para o ZMPT101B, não é tão simples, não domino o python então declarar sensor.py está tanto difícil para mim. estou me baseando em um projeto do https://github.com/sourabhjaiswal/zmpt101b