esphome/feature-requests #1392: [DSMR] add field to sensor.py -> electricity_tariff

zuidwijk, 2021-09-01:
**Describe the problem you have/What new integration you would like**

I had some request from customers who would like to see a sensor for which tariff is used (high or low).

It is here defined in my old code:
https://github.com/zuidwijk/dsmr/blob/main/components/dsmr/fields.h#L228

But never put it to sensor.py as I didn't used it at that time.

It is not defined here:
https://github.com/esphome/esphome/blob/dev/esphome/components/dsmr/sensor.py

Please add this to sensor.py so customers can use it.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

It is field `0-0:96.14.0(0002)`

I assume that high tariff is 1 and low tariff is 2
It's a string with size 4 according to the specifications .

StijnVdd, 2021-10-31:
vote

mariorottjers, 2025-04-14:
And another vote; Please include electricity_tariff