esphome/feature-requests #1393: use psram for displays when possible

geiseri, 2021-09-02:
now that `esphome::new_buffer(...)  [helpers.h#L327](https://github.com/esphome/esphome/blob/dev/esphome/core/helpers.h#L327) exists can it  be used by default for allocating display memory at [display_buffer.cpp#L17](https://github.com/esphome/esphome/blob/dev/esphome/components/display/display_buffer.cpp#L17).  This is already done manually for inkplate6 but it would help out on the M5 devices because their displays suck up too much ram and break Bluetooth any other integrations in non-obvious ways. See: https://github.com/esphome/issues/issues/2045 (one of many)

geiseri, 2022-10-14:
This has since been resolved.

closed 2022-10-14 by geiseri

nielsnl68, 2022-10-15:
How did you resolve it?