esphome/feature-requests #1396: Add support for ANT BMS

lucasimons, 2021-09-02:
**Describe the problem you have/What new integration you would like**

Add support for ANT BMS over UART or ble uart
**Please describe your use case for this integration and alternatives you've tried:**

Now i use a python script to read the data and send to mqtt on a rpi4
**Additional context**

https://github.com/juamiso/ANT_BMS
https://github.com/Sgw32/BMSCtl

SeByDocKy, 2022-01-01:
I got a custom sensor for the ANTBMS 2020 ... Unfortunatly they changed the protocol for the 2021 model :(

lucasimons, 2022-01-01:
Can you explain to me how you did it? did you connect to the display port? I currently use a script and connect to the bms via bluetooth

SeByDocKy, 2022-01-01:
Yes on the display port.... it's an UART ... so basic wiring, ANTBMS: (GND,TX,RX) <=> ESP32 : (GND,RX,TX)

lucasimons, 2022-01-01:
Have you used esphome?  Do you also have the option to switch the mosfet?

SeByDocKy, 2022-01-01:
Yes ESPhome with a custom sensor : https://drive.google.com/file/d/1dwPWgargfSiYi-KMFOFXDH9V1jTvhBTh/view?usp=sharing

You can play with it... Tuned for a 16S battery

lucasimons, 2022-01-01:
thank you very much, how did you connect the cables? i.e. you only connected gnd to gnd, rx to tx (gpio01), tx to rx (gpio03)

SeByDocKy, 2022-01-01:
right

lucasimons, 2022-01-02:
How does it work when you connect the display cable to the esp, the bluetooth still works?

lumiror, 2022-01-04:
> Áno ESPhome s vlastným senzorom: https://drive.google.com/file/d/1dwPWgargfSiYi-KMFOFXDH9V1jTvhBTh/view?usp=sharing
> 
> Dá sa s tým hrať... Vyladené na 16S batériu

You can describe it in more detail, I'm not an experienced user, I would need a little more detailed explanation if possible. Thank you

lumiror, 2022-01-04:
> Yes ESPhome with a custom sensor : https://drive.google.com/file/d/1dwPWgargfSiYi-KMFOFXDH9V1jTvhBTh/view?usp=sharing
> 
> You can play with it... Tuned for a 16S battery

You can describe it in more detail, I'm not an experienced user, I would need a little more detailed explanation if possible. Thank you

SeByDocKy, 2022-01-04:
Juste install the yaml and .h files into your /config/Esphome directory. Edit in Esphome the yaml file and change your WiFi informations... Be sure to connect the TX & RX to the pins specified in the top of the .h file... Than compile and install the file on your ESP32.

lucasimons, 2022-01-20:
I received the batteries so now I would like to use the bms if I'm not mistaken this connector I have to use it to power the two wires to turn on the bms after connecting them in short circuit, instead the other two wires are rx and tx right?  Can you send some pictures of the links?  I thank you in advance

lumiror, 2022-01-23:
> Stačí nainštalovať súbory yaml a .h do vášho adresára /config/Esphome. Upravte v Esphome súbor yaml a zmeňte informácie o WiFi... Uistite sa, že ste TX & RX pripojili ku kolíkom uvedeným v hornej časti súboru .h... Potom súbor skompilujte a nainštalujte do svojho ESP32.

Thank you for the answer, it is possible to use the module WeMos D1 Mini, ie ESP8266

SeByDocKy, 2022-01-23:
As I wrote from the beginnin the code I wrote sometimes ago was for the ANTbms model 2020.... Not sure it will work with the 2021 version as you have since the Appz changed between.... I guess they changed something in the protocol. But yes it's possible to use a simple WeMos D1 mini

lumiror, 2022-01-24:
I don't know what I'm doing wrong but if I add an "includes" command I get an error, can you please help me how to fix it? I created a new file "antbms_custom_sensor.h" is that ok? Thanks.

SeByDocKy, 2022-01-24:
The include part must be in the esphome: section like

```esphome:
  name: antbms
  platform: ESP32
  board: esp32dev
  includes:
    - antbms_custom_sensor.h```

lumiror, 2022-01-24:
Thank you for the quick reply it helped. Can you please help me again? I got a SERIAL error, I changed the SERIAL2 to SERIAL1 error then only to SERIAL and also the error so I don't know how to fix it.
[image]

lumiror, 2022-01-30:
So I managed to compile and upload to esp32 because esp 2886 didn't work. But now I have a problem, in homeassistant I see no data for entities. In antbms_custom_sensor.h is RX at 17 and TX at 16, is that correct? It should not be the other way around. Can there be a problem with that? I also tried TX on Tx RX on RX but I still don't see any data just unknown. Plus contact from BMS should not be connected? Because if I connect a display without a plus contact, the BMS will not connect to the display. Can you please help where there may be a mistake, maybe I'm close to making it work, I'd be very grateful. That's how I connected it.

[image]

HaraldRuppert, 2022-02-02:
Hi,
same for me. I've also just tried to connect it the way it's described here: https://github.com/imval/AntBMS/issues/1
@lumiror are you using it for 8S or 16S? I've also tried to port it to ESP8266 but even with my ESP32 and unmodified code, ​it doesn't read anything.

Look  at the pinout of your  ANT-BMS LCD btw. the colors for the wires don´t make logical sense.
​For me, green = ground, red = VCC, yellow = RX and black = TX.

But I also struggle to read anything on any wire with a multimeter so I wonder if they even work.

lumiror, 2022-02-02:
Hi, I use 16S exactly as the author. Unfortunately, I still don't get any data. Maybe if the author gave a photo of which version of BMS he would use, we would move somewhere. I had those cables just like you, but before the assembly I swapped them according to the standard. Do you have a program that works via bluetooth?

lucasimons, 2022-02-03:
hello, but you have turned on the bms, that is through bluetooh it works for you? I always have to receive the cells then I try if it works, as I asked in this discussion I did not know if I had to connect the black and red cables together because in the manual it is written that you have to apply a voltage of 3.2v between the two cables and then connect them later together, what procedure did you use?

lumiror, 2022-02-03:
Via bluetooth it works with the original application, but only locally and limited bluetooth range. I don't understand the question, what red and black cable do you want to connect?

SeByDocKy, 2022-02-16:
> So I managed to make it work. I just have a little problem. The program freezes quite often, so I want to ask someone who uses it, if it happens to you and if so, how often. Thanks

try to remove the date_baud: 0 in the logger and maybe you will have to adjust some timing in the .h

lumiror, 2022-02-16:
Could you be more detailed please? Where can I find timing in the .h so which line is it. Alternatively, as I delete date_baud: 0 in the logger. I am a beginner and this information will help me a lot.

lucasimons, 2022-02-20:
Hi, unfortunately I have not yet been able to test the communication between bms and esp but I have seen that new repositories have been created for the various bms and integrated with esphome like this for jk bms

https://github.com/syssi/esphome-jk-bms

, it would be interesting to create a repo for ant bms

lucasimons, 2022-02-21:
Hi, if you want to join to test and implement ant bms on esphome you can write here

lumiror, 2022-02-21:
Just today I finished testing a new version from 2021 which also works. I am very satisfied . I have an old and a new version of BMS and they both work via the display port. If you have a display for your BMS and displays the data it is highly likely that it will work for you as well. Then just tune the stability (this must be done by someone who knows how to program, unfortunately I don't know) and it will be a great version for ANT BMS. Thanks to the author for sharing.
[image]

lucasimons, 2022-02-21:
I also have two bms one 100a old and one new 300a purchased in October 2021, so I think I am in the same condition as you, have you made any changes to the code? Can you explain better?

SeByDocKy, 2022-02-21:
> Just today I finished testing a new version from 2021 which also works. I am very satisfied . I have an old and a new version of BMS and they both work via the display port. If you have a display for yourANT BMS and displays the data it is highly likely that it will work for you as well. Then just tune the stability (this must be done by someone who knows how to program, unfortunately I don't know) and it will be a great version for ANT BMS. Thanks to the author for sharing. [image]

It's me :)

lumiror, 2022-02-21:
> >
> It's me :)

I know

[image]

lumiror, 2022-02-21:
> I also have two bms one 100a old and one new 300a purchased in October 2021, so I think I am in the same condition as you, have you made any changes to the code? Can you explain better?

I didn't change anything, I just copied it to ESPHome and ESP32 as described by SeByDocKy.

SeByDocKy, 2022-02-21:
> > I also have two bms one 100a old and one new 300a purchased in October 2021, so I think I am in the same condition as you, have you made any changes to the code? Can you explain better?
> 
> I didn't change anything, I just copied it to ESPHome and ESP32 as described by SeByDocKy.

Glad to know that my code still work with the antbms 2021.... Some tested but was not working. I went directlly to the JKBMS when I saw the amazing @syssi integration

lumiror, 2022-02-21:
As I wrote, I had to connect plus contact otherwise it also didn't work for me.

lucasimons, 2022-02-21:
> > I also have two bms one 100a old and one new 300a purchased in October 2021, so I think I am in the same condition as you, have you made any changes to the code? Can you explain better?
> 
> I didn't change anything, I just copied it to ESPHome and ESP32 as described by SeByDocKy.
> 
> **Note:** I also had to connect a plus contact to 3.3V on my BMS. I used a 3.3V output from ESP32. :)

how did you connect the esp32 to the bms? do you have a picture?

lucasimons, 2022-02-21:
> Glad to know that my code still work with the antbms 2021.... Some tested but was not working. I went directlly to the JKBMS when I saw the amazing @syssi integration

I also saw the various syssi integrations and in fact I asked to @syssi if it was possible to merge what you wrote and integrate it on esphome, seen the results of @lumiror  I believe that there are no problems

lumiror, 2022-02-21:
As I described. Just be careful, I swapped cables. You have them connected differently by color
[image]
.

syssi, 2022-02-21:
@SeByDocKy You did a great job building the custom sensor setup! I've converted your findings/knowledge into a custom component and would be happy about some testers. Please give a try:

https://github.com/syssi/esphome-ant-bms/
https://github.com/syssi/esphome-ant-bms/blob/main/esp32-example.yaml

If it doesn't work out of the box please create an issue and provide some log output. The debug output of the `uart` component is enabled per default at the moment to log as much as possible (the serial traffic).

SeByDocKy, 2022-02-21:
> @SeByDocKy You did a great job building the custom sensor setup! I've converted your findings/knowledge into a custom component and would be happy about some testers. Please give a try: [...]

WHaooohhhh so great :) Incredible. I know several person with the AntBMS 2021 (I have one too)... so  we will have feedbacks. Little out of topic. Why you don't ask the official integration of your JKBMS into ESPhome ?

syssi, 2022-02-21:
Off-topic answer: Because of some "technical debt". The JK-BMS component uses a modified version of the [modbus component](https://github.com/esphome/esphome/tree/dev/esphome/components/modbus) component. The successor of the `modbus` component is the [modbus_controller](https://github.com/esphome/esphome/tree/dev/esphome/components/modbus_controller) (a great component with important features f.e. send individual frames + queueing of commands/requests + expect responses). To control the BMS command queueing is required. The generic `modbus` component cannot be extended easily without becoming device specific. IMO the JK-BMS component shouldn't merged into ESPHome as long it doesn't use the `modbus_controller` as base.

lucasimons, 2022-02-21:
> platform: ESP32
>   board: esp32dev
I tried to use the custom component of @SeByDocKy , but unfortunately it doesn't work with esp8266 and with esp32 I have these disconnections
[image]
[image]

lucasimons, 2022-02-21:
This is my configuration, I have also tried to connect the red wire with the 3.3v but i don't show anything on home assistant

lumiror, 2022-02-22:
Your BMS is different. Do you have a display? Does it show you data from BMS? As you have it connected, take a picture.

lucasimons, 2022-02-22:
> Your BMS is different. Do you have a display? Does it show you data from BMS? As you have it connected, take a picture.

I have two bms both without display, I soldered some wires because I did not have the display connector however starting from the left it says on the pcb positive, gnd, B, A, assuming that the last two pins are for the display

lumiror, 2022-02-22:
I have it connected like this:
positive to output  3.3v ESP32
gnd = gnd
A = RX
B = TX

lumiror, 2022-02-22:
still check if you have correctly assigned pins, in the program it is:
#define RXD2 17
#define TXD2 16
but your ESP32 has the other way around

lucasimons, 2022-02-22:
I connected the left pin to 3.3 the pin 2 to gnd then the white cable to 16 and the right cable to 17, so I have no instability problems, but it gives an error trame
[image]

lucasimons, 2022-02-22:
the original display cable have this color red, black, yellow, green 

red to 3.3v
gnd to gnd
yellow ( white in my pic) to tx2 
green  ( yellow in my pic) to rx2

lumiror, 2022-02-22:
For me, the flood color of cables:
  red = VCC
green = GND
yellow = A (RX)
black = B (TX)

lumiror, 2022-02-22:
Do you see the data in the APP via bluetooth? If so, try to change RX and TX or disconnect the VCC 3,3V from ESP 32. You have to try several options and see what it does.

lucasimons, 2022-02-22:
I think I got confused with the cables ... Anyway, do you know what kind of connector is the one of the display?

lumiror, 2022-02-22:
I don't know exactly what type it is, but it's this:

[image]

SeByDocKy, 2022-02-22:
Should be JST XH2.0 4P or JST/microMolex 1.25mm 4p

lucasimons, 2022-02-23:
I tried but through the custom components it does not work and there are always problems with the esp 32 that I have and unfortunately I do not know the reason but it does not work on esp8266, I had reversed the cables because the black was not gnd but tx

lumiror, 2022-02-23:
This is not a BMS mainboard, but a display

syssi, 2022-02-23:
Good point. It helps nevertheless to identify `VCC` and `GND`. `TX` and `RX` will have to be tried out.

lumiror, 2022-02-23:
> I tried but through the custom components it does not work and there are always problems with the esp 32 that I have and unfortunately I do not know the reason but it does not work on esp8266, I had reversed the cables because the black was not gnd but tx

ESP8266 don't even try it doesn't work, but as Syssi writes take photos as you have it connected. But good quality, focused so we can see the details

lumiror, 2022-02-23:
..... something to motivate [image]

[image]

lumiror, 2022-02-23:
take yet photo of ESP32

lucasimons, 2022-02-23:
[image]

lumiror, 2022-02-23:
It is not visible to which pins you have the green and yellow cable connected

[5 more replies]

closed 2022-07-01 by nagyrobi