esphome/feature-requests #1397: Zigbee/Thread support on the new upcoming ESP32-H2 variant?

levihb, 2021-09-03:
**Describe the problem you have/What new integration you would like**

A new variant of the ESP32 [was announced last month](https://www.espressif.com/en/news/ESP32_H2), the ESP32-H2. It supports 802.15.4 and therefore Zigbee and Thread, as well as BLE, and now runs on a RISC-V architecture which should be a lot more efficient. It also still supports WiFi.

This looks super interesting because it will finally give home users a way to easily build Zigbee/Thread versions instead of WiFi ones. I think ESPHome supporting this would be a brilliant step forward.

**Please describe your use case for this integration and alternatives you've tried:**

There are no alternatives to ESPHome for Zigbee as far as I know? Or really any simple development environments for home users.

**Additional context**

Of course the H2 isn't out yet. But I think it'd be good to get a discussion on this going before it is released. Zigbee is definitely a lot more friendly to larger home installations in many ways.

Support for battery operated end Zigbee devices would also be a huge leap forward. And I would think that should be possible on the new H2.

oxan, 2021-09-06:
The big open question here is whether some group will port a Zigbee stack to the H2, as that's likely a too big task to do for ESPHome by itself.

levihb, 2021-09-20:
> The big open question here is whether some group will port a Zigbee stack to the H2, as that's likely a too big task to do for ESPHome by itself.

I [asked on the ESP forums](https://www.esp32.com/viewtopic.php?f=2&t=23147). Looks like they haven't released any information publicly yet.

But there will be an OpenThread implementation available. What about potential support for that?

levihb, 2021-09-25:
@oxan Excellent news. [They have confirmed that there will be a Zigbee stack available!](https://www.esp32.com/viewtopic.php?f=2&t=23147&sid=43c0590c7419e9dc2e0bc8b6c43ceeb1#p83948)

Hopefully this means we can have a Zigbee ESPHome.

DelusionalAI, 2022-01-11:
With thread being IP based, would it be possible to connect it to a ESPHome thread device to a thread-border router and then control it over existing tools like MQTT and the Esphome API?  
I believe the only "issue" would be that Thread is IPv6 only, which [is being worked on](https://github.com/esphome/esphome/pull/2953)

Controlling my devices over the existing APIs but without worrying about wifi would be amazing, and then you wouldn't be limited to the zigbee protocols.

oxan, 2022-01-11:
I looked a tiny bit into this, and getting the API to run on Thread seems like it'll be fairly straighforward. Once devices start becoming available that's probably something we'll work to support.

As for Zigbee, examples using the Z-Boss 3.0 stack have started appearing in the IDF repository last week, so I bet that's what they're going with. There's actually an example of how to use an ESP as a lightbulb. It doesn't seem like it'll be too hard to integrate that into ESPHome as well, in case anyone is interested. I'm happy to help people implementing this, but probably won't work on it myself.

Drizzt321, 2022-01-17:
Really interested in this myself! Would love to create a few battery sipping things for myself, particularly Zigbee 3.0.

Hedda, 2022-02-21:
> With thread being IP based, would it be possible to connect it to a ESPHome thread device to a thread-border router and then control it over existing tools like MQTT and the Esphome AP

That could indeed be possible if Espressif releases a CLI/API for Zigbee NCP or an SDK for RCP, and that is something that Tasmota already feature support for but only for Silicon Labs EZSP and Texas Instruments ZNP, see: https://tasmota.github.io/docs/Zigbee/

There is by the way a related similar request here for Silabs EZSP (EmberZNet Serial Protocol) -> https://github.com/esphome/feature-requests/issues/681

There is also a separate request for just pass-through serial stream server (ser2net tunnel) support for Zigbee MCU here -> https://github.com/esphome/feature-requests/issues/688

kaystrobach, 2022-03-26:
Related?

https://community.home-assistant.io/t/announcement-esphome-running-on-adafruit-52840-over-thread-mesh-networking-using-zephyrrtos/363712

HarvsG, 2022-03-27:
I would very happy to contribute to the esphome team getting an engineering sample ASAP (if it is a matter of $) - Apparently they are hoping to get them out in May 2022 [according to this reddit post](https://www.reddit.com/r/esp32/comments/pjc28v/when_will_the_esp32h2_be_available_for_purchase/)

Hedda, 2022-03-28:
Espressif now looks to have released ESP ZBOSS 3.0 binary libraries supporting the Zboss Zigbee 3.0 stack for ESP32 series chips:

https://components.espressif.com/component/espressif/esp-zboss-lib

https://www.esp32.com/viewtopic.php?t=24624

FYI, Espressif has joined "ZBOSS Open Initiative (ZOI)" so chances are they will base is Zigbee stack on ZOI's ZBOSS stack by DRS:

https://dsr-iot.com/news/78

https://www.prweb.com/releases/espressif_joins_zboss_open_initiative_zoi_a_community_for_royalty_free_zigbee_pro_stack_software/prweb18263163.htm

ZOI (ZBOSS Open Initiative) ZBOSS stack look to be available as royalty-free for ZOI members like but not as open source code:

https://dsr-zoi.com/frequently-asked-questions/

https://www.prweb.com/releases/announcing_launch_of_zboss_open_initiative_zoi_for_royalty_free_zigbee_pro_stack_software/prweb16814352.htm

https://dsr-zboss.com/#!/updates/dsr-joins-cmgc-partnership-s2link

All support for it will need be handled though Espressif as implementer and not directly to ZOI or DRS.

There now also looks to be Zigbee examples in Espressif’s IoT Development Framework (ESP-IDF) but no news on SDK or docs:

There now looks to be some Zigbee examples in Espressif’s IoT Development Framework (ESP-IDF):

https://components.espressif.com/component/espressif/esp-zboss-lib

https://www.espressif.com/en/products/sdks/esp-idf

https://github.com/espressif/esp-idf

https://github.com/espressif/esp-idf/tree/master/examples/zigbee

Those examples include; Zigbee coordinator ("light coordinator"), "Zigbee RCP (radio co-processor"), and "Zigbee gateway":

https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample/light_coordinator

"**_light_coordinator is a light coordinator example demonstrating Zigbee Coordinator role. It provides a formation of the Zigbee network. It runs on an 802.15.4 SoC like ESP32-H2. For more details to see the example readme file._**"

https://github.com/espressif/esp-idf/tree/master/examples/zigbee/esp_zigbee_rcp

https://github.com/espressif/esp-idf/tree/master/examples/zigbee/esp_zigbee_gateway

Wondering if "RPC" works similar to Silicon Labs RPC which works like a dumb radio requiring external ESP ZBOSS 3.0 libraries?

https://components.espressif.com/component/espressif/esp-zboss-lib

"**_This test code shows how to configure Zigbee rcp (radio co-processor) device. Rcp doesn't function alone, it needs to work together with Zigbee gateway (see esp_zigbee_gateway example)_**"

"**_After rcp starts up, it will send its own MAC ieee address and Zigbee stack version number to the Zigbee gateway and start working together with Zigbee gateway via UART communication_**"

ESP32-H2 has apparently now also already been certified as a “Zigbee Compliant Platform” by CSA:

https://www.espressif.com/en/news/ESP32-H2_Thread_and_Zigbee_certified

https://www.espressif.com/sites/default/files/ESP32-H2%20Zigbee%20Compliant%20Platform%20Certification.pdf

_**Type of Device:  Zigbee Compliant Platform
Zigbee PRO Feature Set (2017)
Manufacturer:  Espressif Systems (Shanghai) Co., Ltd.
Model Identification:  ESP32-H2
Firmware Version: V1.0
Hardware Version: V1.0
Certification Date: October 20, 2021
Certification ID Number: ZIG21030ZCP27315-24**_

PS: ESP32-H2 has also passed "Thread Certified Component” certification and will surely also be prioritizing Matter certification: 

https://youtu.be/bS9Ch7k_GK0

They also say Espressif’s IoT Development Framework (ESP-IDF) feature a Thread SDK and that OpenThread is now supported:

https://github.com/espressif/esp-idf/tree/master/examples/openthread

HarvsG, 2022-03-29:
@Hedda thanks for that information dump. Could I be bold and ask what the implications for esphome are? Sounds like many of the software stacks are already quite mature. Do you think that the team have enough to work with or do they need the SDK first?

Hedda, 2022-09-27:
FYI, ESP-IDF (Espressif IoT Development Framework) look to now have posted an OpenThread RCP (Radio Co-Processor) example:

https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp

I think that is needed for ESPHome to support OpenThread and Matter over Thread on ESP32-H2 with its IEEE 802.15.4 radio?

https://www.espressif.com/en/news/ESP32_H2

At least if that could be built so firmware provides OpenThread RCP (Radio Co-Processor) mode with SPI interface over the Spinel protocol (spinel+hdlc+uart serial protocol) and tunnel that serial stream to agners OTBR addon (OpenThread Border Router add-on via OpenThread adapter in RCP mode) from Home Assistant addons-development repository for testing Thread/Matter support inside Home Assistant:

https://community.home-assistant.io/t/chip-matter-support-including-thread/305633/88

https://github.com/home-assistant/addons-development/tree/master/openthread_border_router

https://github.com/openthread/ot-br-posix/

https://openthread.io/platforms/co-processor

https://github.com/agners

https://community.home-assistant.io/u/agners/summary

The main reason for OpenThread RCP (Radio Co-Processor) support in an "RPC" firmware with OTBR (OpenThread Border Router) support will make ESP32-H2 with its IEEE 802.15.4 radio compatible with the upcoming Thread stack / IEEE 802.15.4 radio based "Matter" (Project CHIP / Connected Home over IP) devices if used in Home Assistant with their other add-ons for the that is also in development. That addon by agners requires that the IEEE 802.15.4 radio have a OpenThread firmware in "RPC" mode instead of the no traditional "NCP" mode of OpenThread stack running on the IEEE 802.15.4 radio:

https://github.com/home-assistant/addons-development/tree/master/chip_controller_repl

https://github.com/home-assistant/addons-development/tree/master/chip_tool

https://github.com/project-chip/connectedhomeip

https://buildwithmatter.com

https://csa-iot.org/all-solutions/matter/

Also having optional OpenThread "NCP" border router firmware would allow users to alternatively use other existing OpenThread applications that use "NCP" mode instead of the newer "RPC" mode which require additional component running on the host. Ex:

https://github.com/openthread/wpantund

Note that as far as I read so far agners has only worked with Silicon Labs based adapter with OpenThread "RPC" firmware for Thread / EEE 802.15.4 radio based Matter (as well as ESP32-C3 based devkit for Matter over WiFi) and that is only because it is a Silabs EFR32MG21 chip based adapter that will ship inside the official Home Assistant Yellow (formerly Home Assistant Amber) hardware as well as the upcoming official Home Assistant SkyConnect USB Stick dongle:

https://github.com/zigpy/zigpy/discussions/894

https://github.com/home-assistant/addons-development/tree/master/silabs-multiprotocol

https://www.home-assistant.io/blog/2021/09/13/home-assistant-yellow/

https://www.crowdsupply.com/nabu-casa/home-assistant-yellow

https://community.home-assistant.io/t/home-assistant-skyconnect-usb-stick-announced-will-be-compatible-with-both-zigbee-and-thread-including-matter-chip-over-thread/433594

Hedda, 2022-09-27:
Noticed now that they also released "ESP" Zigbee RCP example too for their ESP32-H2 DevKitC (ESP32-H2-DevKitC-1 V2.1):

https://github.com/espressif/esp-idf/tree/master/examples/zigbee/esp_zigbee_rcp

That could be used in combination with their gateway example:

https://github.com/espressif/esp-idf/tree/master/examples/zigbee/esp_zigbee_gateway

Hedda, 2023-01-31:
**ESP32-C6** SoC, module and development board has now been launched with WiFi 6, BLE 5.0, and 802.15.4, ink. Zigbee and OpenThread!

https://www.espressif.com.cn/en/news/ESP32-C6_Available

https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/index.html

https://github.com/espressif/esp-idf/issues/10423

Espressif launched the ESP32-C6 WiFi 6, Bluetooth 5.0 LE, and 802.15.4 IoT microcontroller by making the ESP32-C6-WROOM-1 module:

https://www.cnx-software.com/2021/04/11/esp32-c6-wifi-6-and-bluetooth-le-risc-v-soc-for-iot-devices/

[image]

[image]

_Espressif Systems has finally launched the ESP32-C6 WiFi 6, Bluetooth 5.0 LE, and 802.15.4 IoT microcontroller by making the ESP32-C6-WROOM-1 module available on [Aliexpress for $14.25](https://fave.co/3VYE3RW) for a pack of 5 modules as well as the ESP32-C6-DevKitC-1 development board that’s [out of stock on the official store](https://fave.co/3vU14ea) at the time of writing. but somehow listed in another store [for $16.22 plus shipping](https://s.click.aliexpress.com/e/_DlvBgFz)._

_Besides the availability announcement, another surprise is the 802.15.4 radio for Zigbee and Thread that was not part of the original [ESP32-C6 announcement](https://www.cnx-software.com/2021/04/11/esp32-c6-wifi-6-and-bluetooth-le-risc-v-soc-for-iot-devices/) in April 2021. The ESP32-C6-DevKitC-1 development board comes with a module with 8MB flash, exposes most of the I/O’s from the ESP32-C3, and comes with two USB-C ports, two buttons, and an RGB LED._

ESP32-C6-DevKitC-1 development board specifications:

- Wireless module  – ESP32-C6-WROOM-1 with
  - ESP32-C6 single-core 32-bit RISC-V clocked up to 160 MHz 320KB ROM, 512KB SRAM, low-power RISC-V core @ up to 20 MHz
  - Wireless – 2.4 GHz WiFi 6, Bluetooth 5.0 LE, 802.15.4 radio for Zigbee 3.0 and Thread. Matter compatible. PCB antenna
8MB SPI flash
- USB – 1x USB Type-C host port, 1x USB Type-C port via UART console
- I/Os – 2x 16-pin headers with GPIO, UART, SPI, ADC, SDIO, USB, I2S, etc…
- Misc – Reset and BOOT buttons, RGB LED, J5 header/jumper for current measurement
- Power Supply (mutually exclusive options)
  - 5V via USB-UART port (recommended)
  - 5V and GND pin headers
  - 3V3 and GND pin headers
- Dimensions – 51.8 x ~20 mm

[image]

[image]

_The new ESP32-C6 module and development board work with the ESP-IDF v5.1, currently under development (see [progress report](https://github.com/espressif/esp-idf/issues/10423)), and some [basic documentation](https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/index.html) can be found on read the docs._

_If you can’t purchase an ESP32-C6 module or board just yet due to limited supplies, don’t despair as more are coming very soon as [Espressif tells](https://www.espressif.com.cn/en/news/ESP32-C6_Available?position=0&list=WQQQ-4A66v8oesBoFKp-rNz9bSQ9dg5w7wc_nNvDt_I) us to expect more products on Adafruit, Akizuki, Digikey, and Mouser._

MiguelAngelLV, 2023-02-18:
The C6 board can be a perfect board for esphome. Wifi, BLE, Zigbee and Thread in only one board!

Hedda, 2023-02-18:
> The C6 board can be a perfect board for esphome. Wifi, BLE, Zigbee and Thread in only one board!

Indeed, and especially think that these could be perfect as a [**"Thread Board Router" (for Matter)**](https://github.com/esphome/feature-requests/issues/1430) or as a "**Zigbee Router**" device.

* The new [Matter standard](https://github.com/esphome/feature-requests/issues/1430)) need many "Thread Board Router" (OpenThread Border Router, a.k.a. OTBR) to connect to a Thread network in order to make it both more redundant and more stable, as a Thread Board Router acts as a network bridge to connect your normal LAN network to the Thread network and it also uses network mesh technology so having more Thread devices will add better range and coverage.

* Zigbee also uses network mesh technology so simply adding more "Zigbee Router" devices to your Zigbee network will improve range and coverage for the whole Zigbee network.

linkedupbits, 2023-03-19:
I am starting to work on at least getting native esphome support for the esp32-c6-DevKitC-1.
I have created a [pull request](https://github.com/espressif/esptool-js/pull/92) to add support to espressive's esptool-js for this board - this library is used by esp-web-tools to flash the board.
I have successfuly uploaded a zigbee espressive sample firmware to the esp32-c6-DevKitC-1 via my local build of esptool-js (but binaries built via esp-idf).
Then I  a local tar.gz of the esp-web-tools library, and dropped it into my local esp-web-tools , and was able to connect to the serial log output, but not able to upload a firmware built by ESPHome.

Where would I find an esp-idf project file for any of the "standard" new device firmwares eg esphome-web-esp32c3.bin or esphome-web-esp32s2.bin for bootstrapping into ESPHOme?
I could then adapt this to build a base new device firmware for this board.

Hedda, 2023-03-20:
> Where would I find an esp-idf project file for any of the "standard" new device firmwares eg esphome-web-esp32c3.bin or esphome-web-esp32s2.bin for bootstrapping into ESPHOme?

Probably best if you start a new dedicated issue for that question 

https://github.com/esphome/feature-requests/

https://github.com/esphome/esp-web-tools/

https://esphome.github.io/esp-web-tools/

https://web.esphome.io/

rotilho, 2023-03-20:
> I am starting to work on at least getting native esphome support for the esp32-c6-DevKitC-1.
> I have created a [pull request](https://github.com/espressif/esptool-js/pull/92) to add support to [...]

You may be interested in this fork. @stintel managed to flash a C6.

https://github.com/stintel/esphome/tree/esp32c6

Hedda, 2023-04-05:
FYI, [01Space](https://github.com/01Space) is sell a tiny ESP32-C6 IoT dev board based on ESP32-C6-WROOM-1 wireless module with two 10-pin headers + USB-C for power:

https://www.aliexpress.us/item/3256805225446433.html

[image]

ESP32-C6 Arduino support will require ESP IDF 5.1 framework which is still under development:

https://github.com/espressif/arduino-esp32/issues/7713

https://github.com/espressif/arduino-esp32/issues/7852

ict-one-nl, 2023-04-16:
> * ZHA integration

Don't forget zigbee2mqtt, this is one of the most mature solutions when it comes to interface with zigbeestuff through mqtt :)

Hedda, 2023-04-17:
> Don't forget zigbee2mqtt

Zigbee2MQTT support is off-topic here as zboss adapter (esp-zboss) support would have to be implemented into the zigbee-herdsman library it depends on and that has nothing to do with ESPHome or Home Assistant, so you would instead have to post a request there instead -> https://github.com/Koenkk/zigbee-herdsman

wombat70, 2023-06-17:
Espressif announced their ESP Thread BR dev platform:
 https://www.hackster.io/news/espressif-launches-a-kit-to-turn-its-esp32-modules-into-certified-thread-border-routers-for-matter-c61db0fce050

Hedda, 2023-06-20:
> Espressif announced their ESP Thread BR dev platform:
> https://www.hackster.io/news/espressif-launches-a-kit-to-turn-its-esp32-modules-into-certified-thread-border-routers-for-matter-c61db0fce050

Nice! That is a new ESP32-H2 + ESP32-C3 combo development board kit and ESP Thread Boarder Router SDK which as been certified as a Thread Border Router (TBR) / OpenThread Border Router (OTBR) which can be used for Matter based on the Thread 1.3 standard, alternatively be used as an as Zigbee gateway/hub or bridge, and they also got an optional Ethernet Sub-Board

https://www.cnx-software.com/2023/06/20/espressif-esp-thread-border-router-board-combines-esp32-h2-esp32-c3-wireless-chips/

https://github.com/espressif/esp-thread-br

https://docs.espressif.com/projects/esp-thread-br/en/latest/esp32/

https://www.aliexpress.us/item/1005005688193617.html

PS: An earlier two-board ESP Thread Border Router solution was previously available as an example project for reference:

https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br

https://openthread.io/guides/border-router/espressif-esp32

[image]

and optional Optional Ethernet Sub Board

[image]

Hedda, 2023-07-01:
Espressif ESP-IDF v5.1 with ESP32-C6 and ESP32-H2 has now been released

https://github.com/espressif/esp-idf/releases/tag/v5.1

HarvsG, 2023-07-01:
Are there anymore requirements before development can start on ESP Home Matter devices?

Hedda, 2023-08-15:
Off-topic but FYI as related; just noticed that grobasoz added some Zigbee RCP and CLI firmware images for ESP32-C6. Note that is not ESPHome build, however, could serve as a partial proof-of-concept in a two-chip setup (similar to [Espressif's new ESP32-H2 + ESP32-C3 combo development board kit](https://www.cnx-software.com/2023/06/20/espressif-esp-thread-border-router-board-combines-esp32-h2-esp32-c3-wireless-chips/) where the ESP32-S3 would be flashed with ESPHome and the ESP32-C6 would only run Zigbee/OpenThread firmware to simply act as a Zigbee/Thread radio connected via serial using CLI or RCP interface). No idea how or if these firmware images work, (but grobasoz is at least well known in Home Assisistant's ZHA community for providing good unofficial Zigbee NCP firmware images for a wide range of Silicon Labs based Zigbee Coordinator adapters): 

* https://github.com/grobasoz/zigbee-firmware/tree/master/ESP32-C6
  * https://github.com/grobasoz/zigbee-firmware/tree/master/ESP32-C6/ZigbeeCLI
  * https://github.com/grobasoz/zigbee-firmware/tree/master/ESP32-C6/Zigbee_RCP

PS: DamKast recently also announced "zigpy-zboss" as an (unofficial and experimental) ZBOSS radio library for zigpy -> https://github.com/kardia-as/zigpy-zboss and that could in theory make zigpy compatible with ESP32-C6/ESP32-H6 if their firmware use the standard ZBOSS API/CLI which if it is the case should make it usable as a remote network-attatched Zigbee Coordinator for Home Assistant's ZHA integration if configure via serial server pass-through as proxy using socat (similar to ser2net) -> https://github.com/zigpy/zigpy/issues/1052 (More zigpy-zboss discussion related to zigpy here -> https://github.com/zigpy/zigpy/issues/394)

Hedda, 2023-08-21:
FYI, [vidplace7](https://github.com/vidplace7)'s & [stintel](https://github.com/stintel)'s pull request with support for ESP32-C2 & ESP32-C6 has now been merged -> https://github.com/esphome/esphome/pull/4377 and there is also an additional PR by stintel adding support for Espressif's [ESP32-C6-DevKitC-1](https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/index.html) too -> https://github.com/esphome/esphome/pull/5289

Note! Those does not include Zigbee or Thread (or Matter) support so I think this feature request is still in place to discuss that.

knallt, 2023-08-22:
Is the Zigbee support with the new ESP-32 e.g. C6 variant already on the official roadmap? It would be a complete gamechanger to use Zigbee ZHA connection instead of Wifi / BT for esphome integrations.

sorryusernameisalreadytaken, 2023-08-28:
> 

I would love to know that too. But I cant find information on that....

Hedda, 2023-08-29:
> Is the Zigbee support with the new ESP-32 e.g. C6 variant already on the official roadmap? It would be a complete gamechanger to use Zigbee ZHA connection instead of Wifi / BT for esphome integrations.
>
>> I would love to know that too. But I cant find information on that.

Questions like that and "+1" posts are better asked in esphome discussions -> https://github.com/esphome/esphome/discussions

I do not think that there is a need to just post new "+1" comments to this issue as they do not actually add anything constructive to this feature request that could perhaps have the opposite effect and annoy developers instead of encouraging as each post in issues generates an e-mail, spamming everyone following this.

If you only want to add a simple "+1" to show that you are interest in this then suggest that you consider just click the symbolic thumbs-up emoji at the bottom of the original post above as that will not generate an e-mail to the followers of this request:

[image]

vidplace7, 2023-10-08:
I've submitted another PR to add *initial* support for ESP32-H2
https://github.com/esphome/esphome/pull/5498

Doing this in hopes that someone smarter than me will develop a Thread/Zigbee implementation in ESPHome, once board support is available.

Hedda, 2023-11-04:
Not sure if any use for ESPHome as an alternative to the [ESP-IDF Zigbee library](https://components.espressif.com/components/espressif/esp-zigbee-lib), but FYI, it looks like @P-R-O-C-H-Y is planning on porting the [ESP-IDF Zigbee library](https://components.espressif.com/components/espressif/esp-zigbee-lib) and ZBOSS libraries to be available as Arduino library/libraries for ESP32:

https://github.com/espressif/arduino-esp32/issues/8807

https://github.com/espressif/esp32-arduino-lib-builder/issues/142

spusuf, 2023-11-17:
> Ideally, we would be able to create a firmware based on ESP-IDF and esp-thread-br

I don't think the br (border router) functionality is required. There are many home assistant compatible border routers, its just a matter of esphome equivalent end device (I don't know if that's the correct Thread terminology, but that's what ZigBee calls it). And the using that endpoint to test which existing/new border routers have functioning discovery and connectivity. Whether or not the communication uses matter or an esp/ZigBee specific communication protocol is up to the Devs. I'm sure the tinkerer won't care as long as it works  

> I'll be following this thread... get it? Thread? 

😭

spusuf, 2023-11-17:
> Whether or not the communication uses matter or an esp/ZigBee specific communication protocol is up to the Devs.

Wait is Matter over ZigBee a thing (or non-matter protocols over Thread) or does the issue of ZigBee and Thread functionality need to be split into two different issues.

Hedda, 2023-11-17:
> Wait is Matter over ZigBee a thing (or non-matter protocols over Thread) or does the issue of ZigBee and Thread functionality need to be split into two different issues.

Thread and Zigbee should definitely be two separate issues, and Matter support should also be its own separate issue too,

To clearify:
* You can not use Zigbee over Thread, (Thread only supports IPv6 so it can support implementation that support IPv6).
 * Matter and Thread are directly related as can use Matter over Thread, (though Matter is also supported over WiFi and Ethernet, without Thread, so Thread support is not a requirement to support Thread, however it is required to support Thread-based Matter devices).
* Matter and Zigbee are only indirectly related, and only a little, as a Matter Bridge can act as a Zigbee gateway to Matter bridge, (i.e. a hub that converts Zigbee devices into virtual Matter devices), but a Matter Controller can also act as a bridge for all other IoT protocols that way, which is why it is only slightly related, so not enough to discuss in this thread. Note that the reverse is not supported, so you can not bridge a native Matter device to a Zigbee network (that is, you can not present a native Matter device on a Zigbee network -> see https://matter-smarthome.de/en/know-how/what-is-a-matter-bridge/ )

Valid paths:

* Matter device with Thread radio <-> Thread network <-> Matter Controller
* * Matter device with WiFi radio <-> Thread network <-> Matter Controller
* Zigbee device -> Zigbee Gateway with a Matter Bridge -> Matter Controller

Hedda, 2023-11-17:
> I don't think the br (border router) functionality is required.

Thread Border Router support is definitely of interest, as a Thread Border Router works similarly to existing Bluetooth Proxy feature.

https://esphome.io/components/bluetooth_proxy.html

Thread radios have very short range, their signals have poor building material penetration, and is also extremely sensitive to EMF/EMI/RMI interference, so having the option to enable native Thread Border Router support in ESP32-C6 with ESPHome would be great as would allow users can then place such ESP32-C6 based devices that are within Ethernet or WiFi reach close to their Thread devices that would otherwise be out of reach from other Thread radios on the same network.

Hedda, 2023-11-17:
> For my part, I just want something that extends the range of ESPHome devices and has a better range than WiFi.

If you are looking for better range than WiFi then Thread and Zigbee is not for you. Suggest you look at LoRa/LoRaWAN instead:

https://github.com/esphome/feature-requests/issues/124

Thread and Zigbee have a much shorter range than WiFi since the physical layer of Thread and Zigbee relies on IEEE 802.15.4 radios that are low-rate low-energy, while in comparison Wi-Fi uses high-rate high-energy radios. Thread and Zigbee transmit low-energy signals that have very poor penetration of building materials and they also send very short messages, (e.i. "turn device xyx on"), so their signals can easily be drowned out in the environment noise from other electronics.

https://en.wikipedia.org/wiki/IEEE_802.15.4

So the benefit with Thread and Zigbee is that they support their own mesh networking so practically any mains-powered Thread and Zigbee device can act as a router that forwards messages within the network that they belong to (e.i. almost all mains-powered Thread and Zigbee devices will act as signal-repeaters/range-extenders within their own network).

Thread also has the bonus feature of supporting multiple "Thread Border Routers" in the same Thread network, compared to Zigbee which only supports a single Zigbee Coordinator on a Zigbee network. So if you can have several Thread Border Router devices on the same network then you both have redundancy and multiple ways to bridge the Matter protocol traffic to other non-Thread networks (e.i. your LAN network via either WiFi or Ethernet).

PS: Off-topic but if you looking for an IoT protocol where each individual device has longer range without going a mesh network with routers/repeaters/extenders then suggest that you instead check out LoRa/LoRaWAN, (or alternatively if ESPHome and ESP32 is not a requirement then maybe check out Z-Wave 700 and Z-Wave 800 which even has a "Z-Wave Long Range" standard).

micfogas, 2023-11-18:
Hedda, great information for people. I think these 2 explanations are a little off though:

> * You can not use Zigbee over Thread, (Thread only supports IPv6 so it can support implementation that support IPv6).

It should be noted that Thread is an alternative to ZigBee and Z-Wave. They are all built on top of 802.15.4 hardware. Bluetooth uses 802.15.1 hardware. And of course WiFi is 802.11 hardware. These IEEE standards define essentially layers 1 and 2 of the OSI model for these devices: the physical layer and the link layer. An 802.15.4 radio should be capable of working with ZigBee or Thread if flashed with appropriate firmware that implements the protocols fully. So ZigBee and Thread cannot run on top of each other, because they essentially operate at the same level. 802.15.4 defines the WPAN standard, or Wireless Personal Area Network. Thread specifically is built using 6LoWPAN, or IPv6 Low Power Wireless Personal Area Network. I'll address this more on the next section.

> * Matter and Thread are directly related as can use Matter over Thread, (though Matter is also supported over WiFi and Ethernet, without Thread, so Thread support is not a requirement to support Thread, however it is required to support Thread-based Matter devices).

Matter is fully compatible with Thread because Matter is built on IPv6. However, Matter is not exclusively 6LoWPAN. Like you said, Matter supports Ethernet, Wifi and more. Matter is specifically build as a Smart Home universal solution. It was originally known as CHIP, or Connected Home over IP. It's designed to be a universal solution and set of standards that allow all smart home devices (everything Tuya, ZigBee, Thread, SmartThings, etc) and hubs (such as Amazon Echo devices, Home Assistant, Google/Nest Hubs, etc) to integrate with each other seamlessly. The idea is to eliminate the need for, for example, an Aqara hub to use Aqara devices, a Hue hub for Philips, etc. Even though Home Assistant has been developed to overcome many of these proprietary systems, remember that Matter is not something designed or developed for Home Assistant. It's universal and meant to allow all smart home devices to integrate seamlessly without the need of something like Home Assistant. That doesn't mean that Home Assistant is pointless; the advanced level of routines, automations, sensor tracking, states, and the ease of implementing your own homemade smart devices with things like ESPHome, are all reasons why Home Assistant is amazing. Not everyone is tech savvy enough or wants the advanced abilities of Home Assistant. Matter is meant to simplify the connected home for everyone, from Home Assistant gurus to Donna with an Echo Dot who wants to add smart bulbs all around her house and has no idea what Tuya, ZigBee, Thread, or SmartThings is. The idea is that you don't have to do a ton of research to ensure compatibility of your smart devices with your hub; you can have an Echo Dot or a Nest Home Mini and buy 5 different smart bulbs, screw them in, and they just work.

The advantage of using Thread over ZigBee is that it natively supports IPv6, which Matter uses. This means you don't need a ZigBee bridge. However, Thread with Matter devices still need to bridge from 802.15.4 to communicate with WiFi and Ethernet Matter devices. Think of WiFi and the need for an Access Point or WiFi router to bridge your wireless packets to Ethernet so that they can communicate with non-wifi connections, such as the internet.

Thread Border Routers are that solution. With ZigBee, you can only have one Coordinator. In fact, you must have exactly 1 coordinator.

Most people using Home Assistant do this through the use of a USB dongle running ConBee, or that have a generic Silicon Labs board that was programmed with a debugger tool, flashed with an older coordinator firmware. I have one of these. I also have an Echo Plus. It's an older Echo device from around 2016 or 2017 I believe. But it has a ZigBee coordinator built into it, which allows it to connect with non-proprietary ZigBee devices like SIGLED. Tuya ZigBee devices require a Tuya ZigBee hub, Aqara ZigBee devices require an Aqara hub, etc. That's what Thread and Matter are meant to eliminate. Thread can have multiple Thread Border Routers and allows for multiple isolated groups of Thread devices to exist. Matter runs on top of Thread, allowing these isolated groups to be part of the same network. They do have to be bridged of course using Wifi or Ethernet. Matter does use a controller as well. Home Assistant has one available. The big tech smart home hubs such as Echo and Nest Hubs take on this role as well. Think of it like your home router that runs the DHCP server, assigns addresses, and is your centralized gateway for all devices in your network to communicate with each other across WiFi and Ethernet, out to the internet. Matter doesn't require the internet, though a Matter network can be accessed through the internet. Your internet router acts as a controller, sort of, for your home network. You could operate a LAN without a central router, but you'd have to assign IP addresses manually, and most Internet providers only give one public IP address, which means that only one device could connect to the Internet at a time.

In an Ethernet network, your home router is generally referred to as an edge router or border router. It sits at the edge or border of your LAN and passes traffic in and out of your LAN to a WAN, which is typically the [... cut ...] sure if they are using Matter on top of it or not. It's not necessary right now with Home Assistant acting as the Border Router by directly controlling the 802.15.4 hardware to interface with Thread end devices. Eventually, Matter will be running on top of all of these things, including Home Assistant, to integrate smart home devices together, and Home Assistant will be the Matter coordinator.

I hope that helps people better understand the differences and what Matter and Thread are, and aren't.

djorr5, 2023-11-21:
Hi all, I have been deep diving lately into the ESP32C6/H2 as I have been playing around with remotely switching off electric fences via a Zigbee power point. Subtle but it works for me. Currently I have a Sonoff Dongle plugged directly into my Pi which is running HA. This acts as the coordinator and then I enrol a Zigbee GPO into HA which I use my companion app when out in the field to turn it off. However now I have another paddock not on my home network so I am thinking to myself weather it might work for me to play around with the new C6 and see if it can be setup as an Zigbee2MQTT device that subscribes back to my home network HA and uses the Zigbee protocol to turn off the GPO. 

I am adequately enough skilled with Kicad to design a board with 2 ESP SoCs to achieve this but what I am unsure about is if this is achievable from a software perspective. I thought I would ask here as I noticed @Hedda and others in this area seem quite well versed in the technology.

- Can I run Zigbee2MQTT on an ESP32 SoC?
- Does the Zigbee support in a C6 support publicly commercial end devices?
- Will ESPHOME support remote instances of Zigbee ESP board?

Hedda, 2023-11-23:
@djorr5 before reading my replies below you should note that this is still only feature requires. There is not yet any Zigbee or Thread support in ESPHome, and as far as as I know, no developer(s) have started working on any of this as of yet.

> * Can I run Zigbee2MQTT on an ESP32 SoC?
> * Does the Zigbee support in a C6 support publicly commercial end devices?
> * Will ESPHOME support remote instances of Zigbee ESP board?

> * Can I run Zigbee2MQTT on an ESP32 SoC?

No, you can not run Zigbee2MQTT natively on ESP32 SoC because it is does not have a small footprint and requires more reoures (partially because it is coded in JavaScript and TypeScript).

If you want a Zigbee gateway that runs natively on a ESP32 SoC then check out the the Zigbee2Tasmota project by Tasmota:

https://tasmota.github.io/docs/Zigbee/

Existing Tasmota developers have however not yet expressed an integrest in native ZBOSS support for ESP32-C6/ESP32-H2, see:

https://github.com/arendst/Tasmota/discussions/20045

> * Does the Zigbee support in a C6 support publicly commercial end devices?

Espressif allows you to use its libraries for/from their esp-zigbee-sdk and esp-zboss-lib, as well as ESP-IDF SDK/libs and esp32-arduino SDK/libs in commercial products if that is what you are asking, (though you really always have to check the license for each and every library and pice of code used in your project before releasing a commercial product). 

https://github.com/espressif/esp-zigbee-sdk

https://github.com/espressif/esp-zboss-lib

ESPHome firmware can also be used in commercial end devices, (but note that support for ESP32-C6 is very new so should not yet be considered "stable"). For reference, Everything Smart Technology makes and sells a couple nice products based on ESPHome:

https://shop.everythingsmart.io

> * Will ESPHOME support remote instances of Zigbee ESP board?

That is what this feature request is partially asking for, this is a feature request, but again, no developer(s) are working on it yet.

> I thought I would ask here as I noticed @Hedda and others in this area seem quite well versed in the technology.

FYI, I might be well read up on the subject but I do not have the coding skills to implement any of the things requested here.

nstadigs, 2023-11-23:
> I also want to add that in my humble opinion, ESPHome is only a good platform if you are targeting Home Assistant users, (that is, ESPHome is not a broad platform that can easily be used in commercial ecosystems if the user does not also have Home Assistant).

I don't agree. I'm using esphome with mqtt, without home-assistant and I still find the declarative approach of esphome is very beneficial.

crbn60, 2023-11-23:
I second that: I do not use HA at all, but heavily use ESPhome to simply
connect MCUs to MQTT.

A.

MiguelAngelLV, 2023-11-24:
Esphome can be used without HA and without connection. You have screen support so you can create easily a device to show data.

For me, before get support Zigbee and Thread, I think is more important get a Matter support over WiFi. So, any device create with EspHome can be used with any domotic system (HA, Google, Alexa....)

AllardKatan, 2023-12-04:
So, just to so I understand things correctly: this H2 board and the C6 board will make it possible to make ESPhome devices that communicate via Zigbee (as end devices or routers) instead of WiFi? That would be great. I don't use any ESPhome devices because my wifi isn't very reliable, all my sensors etc communicate through Zigbee. But I've looked at ESPhome before and it looks like a nice ecosystem. 
But apparently it's still a long way to go until I can expect to buy one of these and pull a few useful projects off the internet to flash onto them? Any rough ball-park estimates of that timeframe would be appreciated. Are there people actively working on it, or is this still waiting for someone to pick it up?

Hedda, 2023-12-04:
> So, just to so I understand things correctly: this H2 board and the C6 board will make it possible to make ESPhome devices that communicate via Zigbee (as end devices or routers) instead of WiFi?

Yes. As proof-of-concept it is possible using Espressif ESP-Zigbee-SDK with example code for ESP-IDF (not via ESPHome though).

https://docs.espressif.com/projects/esp-zigbee-sdk/en/latest/esp32/introduction.html

https://github.com/espressif/esp-zigbee-sdk

> Any rough ball-park estimates of that timeframe would be appreciated. Are there people actively working on it, or is this still waiting for someone to pick it up?

As far as I know no one is working on it for ESPHome as of yet, (at least not what I have seen). This is only a feature request.

micfogas, 2023-12-04:
> As far as I know no one is working on it for ESPHome as of yet, (at least not what I have seen so far). This is only a feature request.

I jumped on the discord a couple weeks ago and started a thread in the #dev channel about it, but I had a medical issue come up that resulted in 2 ER visits, a hospital admission, and 2 surgeries. I'm still recovering from that and my energy isn't very much still, but I want to focus on it again soon. That is, implementing 802.15.4 radio support for ESPHome. I have the Espressif OTBR Gateway/ZigBee board that has an H2 and an S3 on it for the host. Each have their own firmware. I haven't been able to get the host firmware to build with the RCP OTA Firmware option checked, but I didn't spend time with it. I probably have to define the location of the RCP firmware (or put it somewhere specific). The RCP (H2) firmware build example comes with ESP-IDF SDK, while the OTBR S3 build example is part of a separate repo called esp-thread-br. I personally built it using Ethernet instead of wifi but I'll focus on WiFi.

AFAICT, the H2 is specifically their 802.15.4 radio transceiver SoC while SoCs like the C6 have an 802.15.4 radio in the SoC. But for the C6, since the WiFi and 802.15.4 radios share the same processor, they can't share processor time, which means if one is active on any given cycle, the other is not. This translates to meaning that you it's a great integrated SoC for providing 802.11 and 802.15.4 as an option on the same board, but trying to use it as both at the same time will give you poor performance with both. It's possible to do but it'd be like listening to 2 songs at once with 1 radio, having to switch the station back and forth. The longer you spend on one, the more you miss on the other. And you don't know what you don't know.

Anyway I'm going to look at integrating the c6 first probably, because that's what most people would be using. The h2 is simply a co-processor that can be used with a host SoC, but that host has to be supported by ESP-IDF. In fact, the C6 requires ESP-IDF. essentially, the necessary action is to integrate ESP-IDF to ESPHome. The rest comes with it. I'll double check again and if no one has started, I'll set up foundation for a pull request. I can't imagine that the project leads haven't at least started on adding IDF, even if it's behind the scenes.

AllardKatan, 2023-12-05:
> I jumped on the discord a couple weeks ago and started a thread in the #dev channel about it, but I had a medical issue come up that resulted in 2 ER visits, a hospital admission, and 2 surgeries. [...]

Thanks for taking the effort and I wish you a speedy recovery!

> This translates to meaning that you it's a great integrated SoC for providing 802.11 and 802.15.4 as an option on the same board, but trying to use it as both at the same time will give you poor performance with both.

I understand that having a device that can communicate with both or translate one to the other might be important for some. For me, I prefer a system where anything with high data rate is on ethernet, and all the small sensors and actuators use Zigbee or (eventually, but not yet) Thread. So there is at least one user out there who is happy with a device that does only 802.15.4
> Anyway I'm going to look at integrating the c6 first probably, because that's what most people would be using. The h2 is simply a co-processor that can be used with a host SoC, but that host has to [...]
Good luck and thanks again!

TheRed-E, 2023-12-05:
> >  So there is at least one user out there who is happy with a device that does only 802.15.4
> 

Also count me in on that! End device support or zigbee router support would greatly benefit the use for mee! Grab myself a ESP32-H2 dev board. So if there is some testing or help requierd, I might jump in to assist!

Tropaion, 2023-12-09:
As someone who already made two zigbee devices with the c6, I wonder, with the state of the current sdk (with very few clusters supported), does it really make sense to invest to work time implement it currently?
As far as I read about the zigee sdk, espressif doesn't intend to implement all cluster on it's own, only those which are requested one the repo. Four/five months ago I requested two clusters I needed for my device, which have finally been implemented about two weeks ago.

tomaszduda23, 2024-02-02:
I've started porting esphome to nRF52840 a few weeks ago https://github.com/esphome/esphome/pull/6075. It can be used with nrf-sdk which has some zigbee examples https://github.com/nrfconnect/sdk-nrf/tree/main/samples/zigbee. I've not looked in zigbee details. Maybe someone would be interested though.

Hedda, 2024-02-06:
> > Espressif now looks to have released ESP ZBOSS 3.0 binary libraries supporting the Zboss Zigbee 3.0 stack for ESP32 series chips
> 
> FYI, indirectly related; you can now use a Nordic Semi nRF52840 dongle as a Zigbee Coordinator inside ZHA via zigpy-zboss:
> 
> https://github.com/kardia-as/zigpy-zboss
> 
> This ZBOSS radio library for zigpy is unofficial and still experimental so testers need to manually it to ZHA in Home Assistant:
> 
> [zigpy/zigpy#394](https://github.com/zigpy/zigpy/issues/394)

FYI, an Espressif developer has now posted an experimental version of zigpy-espzb available in this repo (but it is not yet on PyPI):

https://github.com/lhespress/zigpy-espzb  (also dependent on https://github.com/zigpy/zigpy-cli/pull/45)

Note! That zigpy-espzb project is not meant to allow you to use ESP32-H2/ESP32-C6 as a Zigbee device (i.e. Zigbee End Device or Zigbee Router devices), instead is will only enable you to use an ESP32-H2/ESP32-C6 flashed with an Zigbee NCP firmware image as a remote Zigbee Coordinator radio adapter (using Serial-over-IP proxy/relay) with a [zigpy](https://github.com/zigpy/zigpy/blob/dev/README.md) based Zigbee gateway application (such as Home Assistant's built-in Zigbee Home Automation integration component -> https://www.home-assistant.io/integrations/zha ).

This is related to this zigpy featre request asking to use ESP32-C6/ESP32-H2 as remote Zigbee Coordinator -> https://github.com/zigpy/zigpy/issues/1052

DNA99, 2024-02-12:
how about the esp32 c6? this has zigbee and is it going to be supported by esphome?

Hedda, 2024-02-12:
> how about the esp32 c6? this has zigbee and is it going to be supported by esphome?

Read or search all the posts in this whole thread. The original post just says ESP32-H2 because only it was released at the time and 
the original poster has not updated the original post. But later when ESP32-C6 was announced it has been mentioned loads of times.

luar123, 2024-03-25:
I have created a simple esphome example of a zigbee sensor: 
https://github.com/luar123/esphome_zb_sensor

The zigbee part is currently hard-coded and included as a custom component including a task. Changing this to an external component with more configuration possibilities would be the next step.

lboue, 2024-04-08:
@Hedda 
We are still awaiting approval of this PR: [Add support for ESP32-H2 and ESP32-C2 #1347](https://github.com/platformio/platform-espressif32/pull/1347)

Hedda, 2024-04-15:
FYI, upcoming ESP32-H4 and ESP32-C5 MCU SoC models announced with 802.15.4 radio for Thread and/or Zigbee (+ Bluetooth).

* The existing ESP32-H2 is a single-core CPU model with 802.15.4 radio and no WiFi.
* The new ESP32-H4 is a dual-core CPU model with 802.15.4 radio and no WiFi, plus optional support for PSRAM.
* The new  ESP32-C5 is a single-core CPU model with 802.15.4 radio and WiFi.
* The existing ESP32-C6 is a dual-core CPU model with 802.15.4 radio and WiFi , plus optional support for PSRAM.

[image]

As I understand, these should in the future eventually all be fully supported by the ESP-IDF framework as well as the ESP-Zigbee-SDK and ESP-Matter-SDK to add more flexibility in choice of model options.

PS: Also announced is the new [ESP32-P4](https://www.cnx-software.com/2023/01/06/espressif-esp32-p4-a-400-mhz-general-purpose-dual-core-risc-v-microcontroller/) which a high-performance MCU SoC without any radios at all, so it is designed to act as a general purpose host MCU that in combination with above SoC to make a 2-SoC solution with Ethernet PHY (like for example a Zigbee Gateway).

https://www.cnx-software.com/2022/09/13/esp-hosted-helps-adds-wifi-connectivity-to-legacy-linux-or-mcu-based-products/

https://docs.espressif.com/projects/esp-at/en/latest/esp32/Get_Started/What_is_ESP-AT.html

Hedda, 2024-05-21:
> I've started porting esphome to nRF52840 a few weeks ago [esphome/esphome#6075](https://github.com/esphome/esphome/pull/6075). It can be used with nrf-sdk which has some zigbee examples https://github.com/nrfconnect/sdk-nrf/tree/main/samples/zigbee. I've not looked in zigbee details. Maybe someone would be interested though.

Looks like tomaszduda23 work on basic support for nRF52840 for ESPHome progressing very nicely and just yesterday he marked his pull request as ready for review and testing by devs/testers:

* https://github.com/esphome/esphome/pull/6075

FYI, I started an end-user discussion about his nRF52840 development in Home Assistant's community forum to not spam here:

https://community.home-assistant.io/t/fyi-a-developer-has-begun-porting-esphome-to-a-nrf5-platform-for-nrf52840/683997

xttlegendapi, 2024-05-26:
Hello everyone! It would be really cool if we can have esphome communication over zigbee with the new esp32 variants like H2 and C6.

xttlegendapi, 2024-05-27:
I want to donate 50 eur to the person(s) that can get this working (with most components working and working with home assistant)

[65 more replies]