esphome/feature-requests #1398: DHT20 temperature and humidity sensor

mcristina422, 2021-09-06:
**Describe the problem you have/What new integration you would like**
Support for the DHT20 temperature and humidity sensor that communicates over I2C.

The sensor is sold here https://www.adafruit.com/product/5183

**Please describe your use case for this integration and alternatives you've tried:**
I'll try implementing a [custom component](https://esphome.io/custom/i2c.html) but I'd be nice to see this in core

**Additional context**

mcristina422, 2021-09-06:
Using some examples I have a basic setup working with a d1_mini
```yaml
esphome:
  name: dht20
  platform: ESP8266
  board: d1_mini
  includes:
  - ahtx0.h
  libraries:
  - "https://github.com/adafruit/Adafruit_AHTX0"

# wifi, ota, api etc

# not needed for some reason? idk
# i2c:
#   - id: bus_a
#     sda: D2
#     scl: D1
#     scan: True
    
sensor:
  - platform: custom
    lambda: |-
      auto aht20 = new AHTX0Sensor();
      App.register_component(aht20);
      return {aht20->temperature_sensor, aht20->humidity_sensor};
    sensors:
    - name: "Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 2
    - name: "Humidity"
      unit_of_measurement: "%"
      accuracy_decimals: 2
```
where the `ahtx0.h` file is 
```cpp
#include "esphome.h"
#include <Wire.h>
#include "Adafruit_AHTX0.h"

#define TAG "ahtx0"

class AHTX0Sensor : public PollingComponent {

  public:
    Adafruit_AHTX0 aht;
    Sensor *temperature_sensor = new Sensor();
    Sensor *humidity_sensor = new Sensor();

    AHTX0Sensor() : PollingComponent(15000) { }

    void setup() override {
      ESP_LOGCONFIG(TAG, "Setting up AHTX0Sensor...");
      if (!aht.begin()) {
        ESP_LOGCONFIG(TAG, "Failed to start AHTX0Sensor...");
      }
      LOG_SENSOR("  ", "Temperature", this->temperature_sensor);
      LOG_SENSOR("  ", "Humidity", this->humidity_sensor);
    }

    void update() override {
      sensors_event_t humidity, temp;
      aht.getEvent(&humidity, &temp);
      temperature_sensor->publish_state(temp.temperature);
      humidity_sensor->publish_state(humidity.relative_humidity);
    }
};
```

mutantmonkey, 2021-09-21:
SparkFun also sells this part as [SEN-18364](https://www.sparkfun.com/products/18364). I was able to get mine working using the [AHT10 sensor component](https://esphome.io/components/sensor/aht10.html). Perhaps the documentation just needs to be updated to indicate that the AHT20 is also supported?

mcristina422, 2021-09-21:
You're right! I missed that when going through the docs, but recompiling using the `aht10` sensor does work!

iAutomator, 2022-02-19:
It would be more clear if dht20 support be finally added so users can specify its name in config instead `aht10`.
@mcristina422 what do you think ?

ekramlich, 2022-03-22:
Using your initial code and related libraries for Adafruit AHT20 beakout board, I was able to get it to compile and install but it is not reading the temperature or humidity output. Any ideas what may be the issue? 

Log output
[16:27:32][D][sensor:124]: 'Temperature': Sending state 0.00000 °F with 2 decimals of accuracy
[16:27:32][D][sensor:124]: 'Humidity': Sending state 0.00000 % with 2 decimals of accuracy
[16:27:40][D][bh1750.sensor:159]: 'BH1750 Illuminance': Got illuminance=11.4lx
[16:27:40][D][sensor:124]: 'BH1750 Illuminance': Sending state 11.43209 lx with 1 decimals of accuracy
[16:27:47][D][sensor:124]: 'Temperature': Sending state 0.00000 °F with 2 decimals of accuracy
[16:27:47][D][sensor:124]: 'Humidity': Sending state 0.00000 % with 2 decimals of accuracy

Code snippet - 

esphome:
  name: esphome1
  platform: ESP8266
  board: esp01_1m
  esp8266_restore_from_flash: True

  includes:
  - ahtx0.h
  libraries:
  - "SPI"
  - "Wire"
  - "https://github.com/adafruit/Adafruit_Sensor"
  - "https://github.com/adafruit/Adafruit_BusIO"
  - "https://github.com/adafruit/Adafruit_AHTX0"

# Enable logging
logger:

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: true
 
sensor:
  - platform: custom
    lambda: |-
      auto aht20 = new AHTX0Sensor();
      App.register_component(aht20);
      return {aht20->temperature_sensor, aht20->humidity_sensor};
    sensors:
    - name: "Temperature"
      unit_of_measurement: °F
      accuracy_decimals: 2
    - name: "Humidity"
      unit_of_measurement: "%"
      accuracy_decimals: 2

ekramlich, 2022-03-22:
Adafruit AHT20 - Temperature & Humidity Sensor Breakout Board - STEMMA QT / Qwiic
[https://www.adafruit.com/product/4566?gclid=Cj0KCQjw5-WRBhCKARIsAAId9Fn7gqy0zm_4i7AVlrELHtuy5NfkjyyXb0z323Sv6rXH3f6swVw4pz0aAqcAEALw_wcB](url)

sumirati, 2022-05-15:
Created PR https://github.com/esphome/esphome-docs/pull/2083

closed 2022-07-01 by nagyrobi