esphome/feature-requests #1399: MQTT componenten :  Add option to not publish only listen

DutchmanNL, 2021-09-06:
**Describe the problem you have/What new integration you would like**
I would like to use the MQTT component as "listen only", maybe an option to turn of publishing states in the settings ?

**Please describe your use case for this integration and alternatives you've tried:**

For text sensors (and usage in displays) i use the MQTT client option as text sensor template which works fine.
All other communication is handled by ESPHome API and want to send data from different machines.

But as a effect of this all current data (states/sensors etc) are also send to the MQTT broker which is unneeded data there.
I would like to have a listen only mode without publishing data to the broker

**Additional context**

rradar, 2021-09-06:
Is mqtt subscribe what you are searching for?

https://esphome.io/components/sensor/mqtt_subscribe.html
https://esphome.io/components/text_sensor/mqtt_subscribe.html

If so please close the issue ;)

DutchmanNL, 2021-09-06:
> Is mqtt subscribe what you are searching for?
> 
> https://esphome.io/components/sensor/mqtt_subscribe.html
> https://esphome.io/components/text_sensor/mqtt_subscribe.html
> 
> If so please close the issue ;)

no, as that subscribe component needs the MQTT part, by adding the mqtt part it will always publish all states and that's exactly what I would like to block 

whitout the mqtt component
[image]
 which connects to the server, I am not able to subscribe on a topic

```
mqtt:
  broker: 10.0.0.2
  username: livingroom
  password: MyMQTTPassword
```

https://esphome.io/components/mqtt.html