esphome/feature-requests #1401: Support Kamstrup Omnipower energy meter

cnrd, 2021-09-07:
**Describe the problem you have/What new integration you would like**

Add support for the [Kamstrup omnipower energy meter](https://www.kamstrup.com/da-dk/elloesninger/intelligente-elmaalere/3-faset-elmaaler) that is used in parts of denmark.

**Please describe your use case for this integration and alternatives you've tried:**

An easier way for people to get data from the energy meter.

**Additional context**

This repo https://github.com/Claustn/esp8266-kamstrup-mqtt already contains a simple implementation that can be deployed on an esp.

The user will need to supply 2 keys that they need to get from the energy company.

Hedda, 2021-09-08:
Wondering if you mean communication via the optical eye head and/or via the HAN-port on those since they have two interfaces?

https://blog.sintef.com/sintefenergy/han-port-smart-meters-ams/

Noted that apparently sme energy meters use DLMS/COSEM protocols and some use a variant of IEC62056-21 Mode D protocol:

https://hanporten.se

https://hanporten.se/svenska/protokollet/

https://github.com/pwitab/iec62056-21

https://hanporten.se/norska/protokollet/

Note! DSMR Slimme Meter integration notes that smart meters with a ‘HAN port’ are not supported by that integration, so I guess it means that it will not work with DSMR in ESPHome using SlimmeLezer hardware since HAN interface is not P1 monitor compatible?

https://www.home-assistant.io/integrations/dsmr/

https://www.home-assistant.io/integrations/p1_monitor/

https://esphome.io/components/sensor/dsmr.html

https://www.zuidwijk.com/product/slimmelezer/

https://github.com/zuidwijk/dsmr/

cnrd, 2021-09-15:
Sorry for not getting back to you earlier.

I was referring to communication on the HAN port.

The primary reason I'm requesting this is to get support for reading data from HAN port based meters from a more mature codebase/solution. All of the current solutions are not really very user friendly, so adding support in esphome would make it more accessible to more users.

I was thinking that configuring it in esphome should expose all of the avaliable data points as sensors in hass (just as what the code in the linked repo does to mqtt), but it would probably be a nicer experience, if hass also understood what it received in a more structured way compared to just dumping a bunch of sensors.

Hedda, 2021-09-16:
Does your specific smart interface module in your Kamstrup Omnipower have an RJ45 interface (a.k.a. Norwegian HAN-port standard) or RJ11 (a.k.a. Swedish HAN-port standard, which I understand in turn is a variant of the P1/H1 port using DSMR protocol)?

Google search says that they at least sell an Kamstrup HAN-module with "HAN-NVE" RJ45 HAN interface with part number 6840004 which according to its specifications listed in the data sheet it looks like the reader needs to comply with MBUS EN-13757-2 standard

https://www.nek.no/wp-content/uploads/2018/10/Kamstrup-HAN-NVE-interface-description_rev_3_1.pdf

https://www.utomhusliv.se/wp-content/uploads/2020/10/Specifikation-f%C3%B6r-HAN-modulen-f%C3%B6r-elm%C3%A4tare-engelska.pdf

Google also says that Aidon AMI system smart meters use the same type of HAN-NVE HAN with RJ45 port with M-Bus interface:

https://www.skekraft.se/wp-content/uploads/2021/03/Aidon_Feature_description_RJ45_HAN_Interface_EN.pdf

https://www.tekniskaverken.se/siteassets/tekniska-verken/elnat/aidonfd-rj45-han-interface-se-v13a.pdf

Again there looks to be some guides for parsing the protocol and getting the data through some kind of M-Bus converter here:

https://www.kode24.no/guider/smart-meter-part-1-getting-the-meter-data/71287300

https://hanporten.se/norska/protokollet/

"They are following the DLMS (Device Language Message Specification) protocol and are sent inside HDLC frames and contains OBIS (Object Identification System) codes that describes the electricity usage. Everything is part of IEC 62056 which is a set of standards for electricity metering data exchange."

https://github.com/pwitab/iec62056-21

PS: If ask for HAN-NVE / RJ45 interface (a.k.a. Norwegian HAN-port standard) then maybe change topic making subject more clear?

jesperrs, 2022-11-27:
This may be made as an extention to the "DSMR Component"...

cnrd, 2026-03-03:
Thanks, I got excited when I saw it added.

Unfortunately the Kamstrup implementation uses two different keys, an encryption key and an authentication key.

I do not really have much time to look into it currently, but if someone feels like it, here is a snippet from Arduino IDE running on an ESP8266 that I have used for years to decrypt the data:
```cpp
bool decrypt(const VectorView& frame) {

  if (frame.size() < headersize + footersize + 12 + 18) {
    Serial.println("Invalid frame size.");
  }

  memcpy(decryptedFrameBuffer, &frame.front(), frame.size());

  uint8_t system_title[8];
  memcpy(system_title, decryptedFrameBuffer + headersize + 2, 8);

  uint8_t initialization_vector[12];
  memcpy(initialization_vector, system_title, 8);
  memcpy(initialization_vector + 8, decryptedFrameBuffer + headersize + 14, 4);

  uint8_t additional_authenticated_data[17];
  memcpy(additional_authenticated_data, decryptedFrameBuffer + headersize + 13, 1);
  memcpy(additional_authenticated_data + 1, authentication_key, 16);

  uint8_t authentication_tag[12];
  memcpy(authentication_tag, decryptedFrameBuffer + headersize + frame.size() - headersize - footersize - 12, 12);

  uint8_t cipher_text[frame.size() - headersize - footersize - 18 - 12];
  memcpy(cipher_text, decryptedFrameBuffer + headersize + 18, frame.size() - headersize - footersize - 12 - 18);

  uint8_t plaintext[sizeof(cipher_text)];

  mbedtls_gcm_init(&m_ctx);
  int success = mbedtls_gcm_setkey(&m_ctx, MBEDTLS_CIPHER_ID_AES, encryption_key, sizeof(encryption_key) * 8);
  if (0 != success) {
    Serial.println("Setkey failed: " + String(success));
    return false;
  }
  success = mbedtls_gcm_auth_decrypt(&m_ctx, sizeof(cipher_text), initialization_vector, sizeof(initialization_vector),
                                     additional_authenticated_data, sizeof(additional_authenticated_data), authentication_tag, sizeof(authentication_tag),
                                     cipher_text, plaintext);
  if (0 != success) {
    Serial.println("authdecrypt failed: " + String(success));
    return false;
  }
  mbedtls_gcm_free(&m_ctx);

  //copy replace encrypted data with decrypted for mbusparser library. Checksum not updated. Hopefully not needed
  memcpy(decryptedFrameBuffer + headersize + 18, plaintext, sizeof(plaintext));
  decryptedFrame = VectorView(decryptedFrameBuffer, frame.size());

  return true;
}
```

Hedda, 2026-03-03:
> I got excited when I saw it added. Unfortunately the Kamstrup implementation uses two different keys, an encryption key and an authentication key.

@SimonFischer04 do you have any thoughts on how to implement the use of two different keys, one encryption key and a separate authentication key, in the new [DLMS Meter component](https://esphome.io/components/sensor/dlms_meter) for ESPHome to make it work with Kamstrup Omnipower Energy meters?