esphome/feature-requests #1402: [REQUEST] Water flow metering reading of Kamstrup smart water meters via IR optical eye header?

Hedda, 2021-09-08:
**Describe the problem you have/What new integration you would like**

Like to get water flow metering from smart water meters with IR eye via hardware similar to SlimmeLezer and Home Assistant Glow.

https://www.mysensors.org/build/ir

http://jheyman.github.io/blog/pages/WirelessWaterMeter/

I believe that a such IR reader adapter could be used with newer Kamstrup water meters with IR optical heads common in Europe.

Specifically, I myself live in Sweden where I think Kamstrup MULTICA 21 / Kamstrup flowIQ 210x ultrasonic water meter is the most common model in normal private residential houses here in Scandinavia (Sweden, Norway, Denmark, Finland, and Island).

https://www.kamstrup.com/en-en/water-solutions/smart-water-meters

https://www.kamstrup.com/en-en/water-solutions/smart-water-meters/multical-21

[image]

[image]

They currently also have Kamstrup flowIQ 2200 "advanced model with acoustic leak detection": 

[image]

[image]

and Kamstrup flowIQ 3100 "Commercial and industrial" model 

https://www.kamstrup.com/en-en/water-solutions/smart-water-meters/flowiq-2200

https://www.kamstrup.com/en-en/water-solutions/smart-water-meters/flowiq-3100

**Please describe your use case for this integration and alternatives you've tried:**

I believe many new water meters offer such two-way bidirectional infrared port (IR receiver and sender header) for communication?

https://www.mysensors.org/build/ir

http://jheyman.github.io/blog/pages/WirelessWaterMeter/

Kamstrup seems to only provide some technical documentation about all their products here but nothing about their protocols:

https://products.kamstrup.com/index.php

Kamstrup apparently call their implementation for Kamstrup Meter Protocol (KMP) and someone has reverse-engineered it here:

https://github.com/bsdphk/PyKamstrup

https://frack.nl/wiki/Stadsverwarming

https://forum.mysensors.org/topic/3525/district-heating-city-heating-stadsverwarming-mysensor-ir-sender-receiver/

https://www.domoticz.com/forum/viewtopic.php?t=11333

As I understand most of those optical probes should work with optical infrared waves be compliant with the international standard IEC 62056-21 (formerly IEC 1107) or ANSI C12.18 communications protocols for energy metering for reading utility meters?

https://en.wikipedia.org/wiki/IEC_62056

https://en.wikipedia.org/wiki/Smart_meter#Protocols

Kamstrup official (and expensive) optical read-out head tools are called 6699-099 (USB) and 699-102 (9F D-sub plug serial)

https://www.termonet.rs/pdf/Kamstrup/Optical%20Read-out%20(IR)%20Head%20-%20Data%20Sheet%20-%20English.pdf

**Additional context**

A hackerspace group in Denmark is selling these IR reading heads for about $14 US (Google translate as test written in Danish) and they wrote a note that using magnets is required as the IR port transmitter / reader eye does not activate unless it sense a magnet:

https://wiki.hal9k.dk/projects/kamstrup

PS: Off-topic but FYI; some if not all of these models of Kamstrup water meters can also be read remotely via Sigfox network RF (Radio Frequency) but that communication traffic is most of the time encrypted, at least if the water meter itself is owned and operated by a utility company or a service provider.

https://www.kamstrup.com/en-en/water-solutions/water-meter-reading

https://www.kamstrup.com/en-en/water-solutions/water-meter-reading/usb-meter-reader

If the case is that the RF radio communication is not encrypted then there look to be projects that can work with that:

https://github.com/adams-okode/kamstrup-integration

https://github.com/weetmuts/wmbusmeters

https://github.com/tobiasrask/wmbus-client

Hedda, 2021-09-08:
This might be related the the request for SML (Smart Message Language) protocol for reading smartmeter? 

https://github.com/esphome/feature-requests/issues/1041

There is a lot collected about workings of SML (a.k.a. "D0 interface") smartmeters with IR two-way here:

https://www.msxfaq.de/sonst/bastelbude/smartmeter_d0_sml.htm

coltju, 2021-11-15:
Check out https://github.com/golles/Home-Assistant-Sensor-MC66C

Im using this for my block heating sensor, which uses the same tech.

evlo, 2022-07-12:
Hi, i'm looking at projects regarding this, either esphome (preferably)

OP did nice summary of the things I also found.

What i understand that IEC 62056-21 is quite widely used in this environment, and sometimes called "S0" (probably something from DIN). I even seems basically UART RS485, just trough optical IR diode(s).
_BTW I remember ages ago IR was used by laptops and cellphones to transfer data._

I think not a lot people know about this and all the projects I found so far about this are quite old/cumbersome or me or at least quite hard to follow.

So I was wondering if i understand it correctly that all need is this
https://www.aliexpress.com/item/1005003727876029.html - connect it using "dupont" cables to nodemcu32 ana i should be able to read the data if speed set to 9600 baud or slower?

evlo, 2022-08-08:
I think the main issue is sending `/?!\r\n` / HEX: `2F3F210D0A` to poll the meter, otherwise maybe the default SML component might work.

evlo, 2022-09-20:
Most likely what my meter spits out is not actually SML, but called IEC62056-21 or sometimes d0 - i think it goes like d0 is the port, IEC62056-21 is the definition of the IR communication, SML is the "extended" protocol and OBIS codes are what each code mean, so even though meter uses IEC62056-21, d0 and OBIS it is most likely not SML, but just IEC62056-21 compliant (which contains basic protocol, but not exactly SML - sml seems to me that it might be IEC62056-21 for multiple meters on the same bus)

This is used by more meters then just water, very similar to SML, but i think separate component is need for easiest implementation. What I wrote in my previous post about using SML component is not true

example:
[image]

SzczepanLeon, 2023-01-04:
@evlo How to get that debug printout?

jakobschou, 2023-01-17:
Hopefully somebody could figure out to get some cheap optical probe like this to work:
https://a.aliexpress.com/_mK4qGxM
so we dont need to solder something ourselves.

It seems that it is possible to do with other Kamstrup meters:
https://github.com/golles/ha-kamstrup_403

Though this setup is using pykamstrup (https://github.com/golles/ha-kamstrup_403/tree/main/custom_components/kamstrup_403) which I dont know if it is the right/same protocol for water meters with IEC62056-21, but maybe it could be changed to pyDLMS (https://github.com/bsdphk/PyDLMS). Here is also some code for the old IEC1107 protocol which maybe could be useful, since the original Kamstrup Optical read-out head seems to be only compliant with this protocol...

Hopefully somebody could fork kamstrup_403 and make it work with Kamstrup Multical 21/FlowIQ 2200 - I just dont have the knowledge to do so.

evlo, 2023-01-17:
maybe take a look at

- https://github.com/evlo/esphome_ZPA-ZE312
- https://github.com/jplitza/esphome_components

using pretty much same thing

cfeenstra1024, 2023-01-21:
If this meter uses the 'Kamstrup Meter Protocol' (KMP) we might be able to use the Kamstrup Multical40x component for ESPHome I've created recently.

If you have the hardware to connect an ESP to the optical interface, I can create a version for ESPHome that can be used to request available parameters. You can then test with your meter. If that works, I can extend the ESPHome component functionality.

I don't have this meter myself, but I do have the Kamstrup Multical 403 (district heating meter) which works well with the created component in ESPHome. I noticed that both meters mention a 'METERTOOL HCM' that can be used with the optical interface in their data sheets. This makes me think they might use the same protocol.

Some info about the component:
- [ESPHome component for Kamstrup multical 40x](https://github.com/cfeenstra1024/esphome/tree/kamstrup_mc40x_temp#readme)
- [Electronics and housing](https://github.com/cfeenstra1024/kamstrup-multical-hardware#readme) I've created for use with the Multical 40x
- [Documentation](https://deploy-preview-2551--esphome.netlify.app/components/sensor/kamstrup_mc40x.html)

onkelbeh, 2023-02-18:
I just printed an adapter to hold an IR adapter on the meter, I am going to give it a try soon.

glance-, 2023-03-21:
I can report that I've had success talking to a flow IQ 2200 using https://github.com/ErikErnst/kamstrup and 1200 baud.

I haven't found which variables the water meter uses yet, but I have read the time / date and serial number from the meter using a ir eye.

jakobschou, 2023-03-21:
> I can report that I've had success talking to a flow IQ 2200 using https://github.com/ErikErnst/kamstrup and 1200 baud.
> 
> 
> 
> I haven't found which variables the water meter uses yet, but I have read the time / date and serial number from the meter using a ir eye.

What IR interface did you use?

glance-, 2023-03-21:
Some random one from flee-bay a couple of years back. The only issue I've found is that the physical form factor of flow IQ 2200 isn't the "standard" one so i had to do some tail then error and when it finally worked apply some tape to keep the ir eye on place. "Normal" devices have a embedded metal plate and some registring features to keep the ir eye in the right place.

copystring, 2023-03-31:
You can send `803f100100444dc00d` which the meter responds for example with `403F1000442804430000215DA390`
In my case the meter has a volume of 8.541m³ which is 215D. To get this number, take 0000215D and convert to decimal (8541). Divide it by 100, and you get the correct volume of 8.541

copystring, 2023-03-31:
Sending `803f100100444dc00d` can be split like this:
| start byte | destination address | CID | requested var | CRC  | end byte |
| 80            | 3f10                          | 01  | 0044                | 4dc0 | 0d           |

on the receiving side:
| start byte | destination address | don't know what this is | requested var | CRC |
| 40 | 3F10 | 0044 | 280443 | 0000215D | A390 |

where 0044 stands for the current meter value

copystring, 2023-03-31:
To calculate the checksum, ignore the start byte. In this case, take 3f10010044. Use CRC-CCITT (XModem) to calculate the CRC.

Maybe this code would work?

```
// Helper function to calculate CRC-16 (Xmodem) checksum for a buffer
function calculateCRC16(buffer) {
  var crc = 0;
  var table = [
    0x0000, 0x1021, 0x2042, 0x3063, 0x4084, 0x50A5, 0x60C6, 0x70E7,
    0x8108, 0x9129, 0xA14A, 0xB16B, 0xC18C, 0xD1AD, 0xE1CE, 0xF1EF
  ];
  
  for (var i = 0; i < buffer.length; i++) {
    var byte = buffer[i];
    crc = ((crc << 8) & 0xFFFF) ^ table[((crc >> 8) ^ byte) & 0xFF];
  }
  
  return crc;
}

// Example usage
var buffer = Buffer.from('803F10010044', 'hex');
var crc = calculateCRC16(buffer);

// Print the CRC checksum in hex format
console.log(crc.toString(16));
```

aquaticus, 2023-04-02:
Hi All,

I created a component for IEC62056-21 meters. It can exchange data with utility meters. Mostly for electricity but also water, thermal and other meters.

It supports both bidirectional and unidirectional communication.
* bidirectional - you need optical interface with IR diode and phototransistor that can send and receive data. This is for modes A,B, and C.
* unidirectional - only IR phototransistor is required. This is for mode D - meter periodically sends data to the interface.

The component does not support binary encoded data (HDLC)..

Hardware required: Optical interface connected to serial port.

IEC 62056-21 term is used for multiple protocols (with the same hardware layer but different data encoding). The component supports meters that provides ASCII encoded data, something like that:
```
1-0:15.8.1(00000009999.567*kWh)
1-0:15.8.2(00000000000.000*kWh)
1-0:15.8.3(00000000000.000*kWh)
1-0:15.8.4(00000000000.000*kWh)
``` 

Detailed description of the component: https://aquaticus.info/iec62056.html

PR to ESPHome: https://github.com/esphome/esphome/pull/4388

copystring, 2023-04-02:
Hi @aquaticus,

does this support only obis codes?

aquaticus, 2023-04-02:
@copystring 
Not sure I understand the question. The component can only interpret information sent as OBIS codes.

copystring, 2023-04-02:
Correct me if I'm wrong, but the way I see it, the Multical 21 does not use OBIS codes.

aquaticus, 2023-04-02:
Just realized Multical 21 probably does not use IEC 62056-21 protocol. I was a bit mislead by the log in one of the comments. In that case the component cannot be used with Multical. And yeah, it seems Multical uses different data model than OBIS.

glance-, 2023-04-10:
> where 0044 stands for the current meter value

I can attest for this. This works also by using readvar from https://github.com/ErikErnst/kamstrup and asking for value 68 (0x44).

280443 is decoded as 0x28 means m^3, 04 means 4 byte value and 43 means how to convert the value, aka 0x40 it's a negative exponent, and 0x03 it should be ^-3 , all according to above referenced code.

glance-, 2023-04-10:
Btw, 74 (0x4A) gives you l/h.

jakobschou, 2023-04-14:
@glance- So to understand - have you figured out how to connect to the meter with the IR-to-USB, and get readings from it yet that can be pushed to etc. Home Assistant?

jakobschou, 2023-04-15:
I have found this fork that should work with the ir-to-usb with HA:

https://github.com/kristianschneider/ha-multical_21

mroek, 2023-04-21:
@glance- I am very interested in your findings, as I also have a flowIQ 2200 in my house (installed by the municipality). The meter sends data using Wireless M-bus, but unfortulately it is encrypted, and the municipality does not want to provide me with the encryption key.

My question to you is whether you know if the data obtained through the optical eye is also going to be encrypted in my case. Do you know whether your meter also sends encrypted data wirelessly? If so, it probably means that the optical data is not encrypted.

Second question is if you have verified that you can poll the meter whenever you want, or if it imposes some limit on the optical interface to preserve battery life?

Third question is if you know whether a magnet is necessary to activate the optical interface on the meter or not. There is no metal plate for the magnet to attach to, so the official Kamstrup optical eye needs a plastic bracket that attaches to the meter to hold the optical eye in place. It could still need a magnet to activate the optical interface, though.

I guess you just used a Linux-machine to run the software you linked to (https://github.com/ErikErnst/kamstrup), but I suppose it should be possible to adapt the necessary parts of the software to run on an ESP to provide a wireless reader of sorts.

mroek, 2023-04-23:
I have a flowIQ2200 (like @glance- ), but there is absolutely no way I can get it to talk to me. I've just DIYed a reading head (3D-printed multiple brackets for the IR leds), and I have added magnets in all kinds of orientations, but the optical interface is just completely dead.

The IR-leds I used was just something I had lying around, and I don't know their peak wavelength, but I doubt that the problem is there. If I point my IR-leds at each other, I will receive a correct echo in a serial terminal, so the circuit (even if just cobbled together on a breadbord) seems fine.

Kamstrup sells a bracket for mounting their optical eye, and due to the design of the meter, the optical eye doesn't really touch the face of the meter, instead it rests against the outer rim. I assume there are light guides in the braket to ensure a good optical connection to the meter. What this also means, is that the magnet sensor in the meter must be located inside the outer rim, and I am fairly certain this is correct. Reason is I found this text in a Kamstrup document:

**_It is possible to toggle through all views by means of the Kamstrup meter reader head, part no. 6699-
099, placed on the optical eye at the meter front. For more details on the meter reader head/optical 
eye, please see chapter 11 Tools and programs. 
The magnet needs to be removed, and applied again on the optical eye, to switch display view._**  

I found a position for the magnet where I could get the meter to change the views, so it seems natural to think that this is also the position used to turn on the optical port. However, still nothing. Very frustrating.

There is also a statement that the polarity of the magnet is important, and I did notice that the above mentioned change of display views only works with the magnet oriented with the south pole towards the meter. This is opposite to what the 62056-21 standard says (for the reading head):

**_Magnetization: axial, north pole directed towards the tariff device._**

It seems unlikely that my meter has a permanently disabled optical port, I can't find any order codes for such an option. And just to make clear, I am sending KMP telegrams, like mentioned by @copystring further up in this discussion.

mroek, 2023-04-23:
Ok, so I finally have success! Turns out that there was indeed a problem with my receiver circuit. It wasn't sensitive enough to trigger on the IR signal from the meter.

I came up with the idea to have only my TX LED connected and sending repeated requests, and then I pointed my phone camera to the TX led of the meter. It showed something was sent, which of course prompted me to increase the sensitivity of the receiver circuit.

After that, I did get sensible data from the meter. Would be nice to have an ID for some more registers, because there are plenty of them. There's a long list in that Kamstrup document I found, but it just names them, there's no telling the address/ID.

mroek, 2023-04-23:
I left my reader polling at 30 seconds interval, and at some point (maybe a couple of hours) the meter stopped responding. :-(

Not sure if there is some internal counter/timer to limit readouts over the optical interface to preserve battery, but it is still bad news, as it means it is not a viable method to monitor the water consumption over time. Unless of course there is an acceptable polling rate that the meter will tolerate. I also noticed that it would shut down the optical interface if it wasn't polled, even if the magnet was left in place.

Will be interesting to see if it starts responding again tomorrow, but it would be really bad design if it doesn't.

mroek, 2023-04-24:
It did start responding again this morning, so it clearly has some battery saver function to prevent polling on the optical interface from depleting the battery. That's annoying, and I have no clue how much polling will be allowed. It does not reset at midnight either, because I tried reading it after midnight, and it was still dead. I checked the internal clock in it this morning, and it was impressively accurate (only about 1 minute off from real time).

It might have some function to count the ON-time of the TX LED (easy enough, given a fixed baud rate, so it can just count the number of transmitted 0 bits and multiply by the bit time), and stop responding after some limit has been reached. However, it must also take time into consideration somehow, and in theory it should be possible to poll continuously if the rate of polling is within the limit. I wish they could just tell us how this works, since reverse-engineering the mechanism takes a lot of time.

It might just start a countdown of 24 hours at the first poll (assuming it has been idle for more than 24 hours), and then when the ON-time limit of the LED has been reached, it will stop responding until the 24 hour countdown has elapsed. That would be a resonable implementation, and if that is indeed how it works, it makes sense to only poll for the values one needs, and not "nice-to-have" values.

aquaticus, 2023-04-24:
@mroek Every battery powered meter applies limits for reading. The producer guarantees device will work for specific time on battery, typically 3-5 years. They compute battery lifespan by assuming max readings per day, so the power consumption is known. 
The number of readings is typically in user manual but guess you have no access to it. For example Maddalena meter I got limits readings to 4 a day.

mroek, 2023-04-24:
@aquaticus Yes, I get that there are (and must be) limits, but it is difficult to get a handle on how this is applied for the flowIQ 2200 that I have. As mentioned, I was able to read multiple registers at a 30 second interval for at least a couple of hours continuously yesterday, but today it seems to have gone silent after just a few reads. And it also seems to deactivate the optical interface even if there is a magnet attached, so the only way to wake it again is to remove and reattach the magnet. That does not work if it has gone all silent, though.

Very frustrating to not have a proper document to refer to. I've asked Kamstrup for some info, but I have not had a reply yet.

mroek, 2023-04-25:
The battery saving function of the flowIQ 2200 basically makes it useless. This morning I did a test where I polled it every 5 minutes, and only requesting one register. It returned 12 values (maybe a couple of more that I didn't catch in my log) before becoming unresponsive. Now, that might mean it is possible to read one value every second hour, but there's another big issue:

The optical interface also times out even if the magnet stays attached, and my tests indicates that this timeout is somewhere between 5 and 10 minutes. That means it is not possible to attach an optical head and then just poll it at a slow enough rate to avoid the polling limit (which is still unknown, just fairly low). The head must be removed and reattached to reactivate the optical interface, which is of course useless. Automating that function is of course also possible (with a servo or possibly an electromagnet), but it really makes for a clumsy solution.

A final note is that I suspect there could even be a global counter for the responses for the optical interface. That would explain why I could read a bunch of values at 30 second intervals the first time I got this working, simply because no readings had been done since the meter was installed a couple of years ago. However, now I have spent my quota, and it will just slowly refill itself.

aquaticus, 2023-04-25:
* Maybe number of readings is limited per week or month. Reading every 30s could block the meter for a couple of days.
* Maybe the meter must be wake up by sending special sequence. Detecting magnetic field could be a one time activation.
* The reader my limit the time it is active, say to 5s, so total transmission time must be less than active time.

The above are just guessing based on other readers I played with.

mroek, 2023-04-25:
If the magnetic field is just left in place, the meter will stop responding even if there are more "reads" available. To get responses again, the magnetic field must be removed and reapplied. However, there could exist a special (and secret) wake-up sequence, of course. It would make some sense if there was, because if service personnel wanted to read for example the logs, it would be dumb if there was no way for them to get any output because there was no quota left due to the user (i.e me) had been reading it until it stopped responding.

There is absolutely no doubt that there is a limit to the available readings, but as of yet it is impossible to know how said limit works. Could be per day, per week, it's anyone's guess, really. Or indeed transmission time. However, as we know, I was able to do a fairly large amount of readings when I first got it working, so clearly there was a number of allowed readings that had accumulated over time. That probably means it is not an amount per week, becuse if it was, then I should have been completely blocked by now. I am not, but the number of allowed readings seems to be very low by now. Probably maybe 10 per day, or something.

One observation I have made that is mildly interesting is that even if the meter has stopped responding, the optical receiver is active (thus a special wakeup sequence could exist). I know this because if you remove and reattach the magnet, you can cycle the display views. The current view will revert after a number of seconds. However, if I send something to the meter in that period, the view reverts immediately. In other words, it does read even if it does not respond.

mdrmdrmdr, 2023-04-26:
After numerous tries with (wrong) ChatGPT suggestions and other websites, I finally could read register x0044 (total volume so far) from my Kamstrup flowIQ 2200 (got it installed just 2 weeks ago) using the code from the [Kamstrup](https://github.com/ErikErnst/kamstrup) repository on GitHub.

But I also see the "stop responding" issue. This indeed makes it quite useless for e.g. once per hour readouts. I really hope that there is some solution to get around this "feature".

Does anybody know if using the official [Kamstrup IR head](https://www.kamstrup.com/en-en/heat-solutions/meters-devices/accessories/optical-readout-head) shows the same behaviour?

mroek, 2023-04-26:
> Does anybody know if using the official [Kamstrup IR head](https://www.kamstrup.com/en-en/heat-solutions/meters-devices/accessories/optical-readout-head) shows the same behaviour?

The optical head doesn't really do anything except convert the data stream to IR light, so I highly doubt that it would make any difference whatsoever. It is possible that the Kamstrup software has some kind of secret sequence to allow it to talk to the meter even after it has stopped responding to normal register reads, but again, that will be a feature in the software, not in the reading head.

Using the official Kamstrup software is probably the only way of finding out if there is a way of getting the meter to respond again, but the software isn't publicly available, and on top of that I suspect it isn't useable without a Kamstrup account.

@mdrmdrmdr Where in the world are you located? Maybe you could get access to the encryption key for your meter, and listen to the radio transmissions from the meter instead?

mdrmdrmdr, 2023-04-26:
> @mdrmdrmdr Where in the world are you located? Maybe you could get access to the encryption key for your meter, and listen to the radio transmissions from the meter instead?

In wonderful Upper Bavaria :-) I doubt that I could get the encryption key. And I'd need to dive into the 868 MHz communication of the flowIQ. I already have a 868 MHz gateway for a differential temperature meter for Homegear, but don't know if this could be used to sniff the Kamstrup radio transmissions.

mdrmdrmdr, 2023-04-26:
I also placed the question regarding the "stop responding" issue  [here](https://github.com/Apollon77/ioBroker.smartmeter/issues/333#issuecomment-1523142249).

mroek, 2023-04-26:
@mdrmdrmdr If you can get access to the encryption key, getting the data from the 868 MHz wireless M-bus is simple enough. You'd just need a suitable radio dongle, and then you could use https://github.com/wmbusmeters/wmbusmeters/ to get the data. I did that, and I can see that my meter sends data, I just can't read it because I haven't got the encryption key.

SzczepanLeon, 2023-04-26:
Here https://forum.arturhome.pl/t/komponent-do-esphome-umozliwiajacy-odczyt-miernikow-wm-bus-i-transmisje-wyniku-do-ha is nice discussion about wM-Bus and ESPHome.

mdrmdrmdr, 2023-04-26:
Thanks, I'll try that. Do you have a suggestion for a good USB radio dongle? The key should be know by the local water people, right? They don't give me the impression that they would understand my question about the key...

SzczepanLeon, 2023-04-26:
> Thanks, I'll try that. Do you have a suggestion for a good USB radio dongle?

Discussion above is about ESP + CC1101. No USB dongles, just ESPHome + external component.

mroek, 2023-04-26:
@mdrmdrmdr Ask them to give you the KEM file for your meter (serial number of your meter is needed). They'd need to log onto the Kamstrup servers to get the file, but they may well refuse. I have had no success so far with my water meter company (the municipality), they just point to GDPR in their refusal.

mroek, 2023-04-26:
As for a dongle, I purchased this (pretty cheap) one: https://www.aliexpress.com/item/1005002623914698.html

Seems to work OK, but again, quite useless without the key.

mdrmdrmdr, 2023-04-26:
Your link responds with "_Sorry, the page you requested can not be found:(_"

mroek, 2023-04-26:
@mdrmdrmdr Very odd. The link works for me, both in Chrome and FF.

mdrmdrmdr, 2023-04-26:
Ah, a second small font comment says: "_Unfortunately, this item is currently unavailable in your location._"

mdrmdrmdr, 2023-04-29:
I'm pretty sure that a wake-up of the IR port of my flowIQ 2200 only is possible by lifting the IR head. I don't think that there is a secret wave-up sequence... I could not find any hint for this. Sad!

Are there IR heads with a USB controllable electrical magnet? ;-)

mroek, 2023-04-29:
> I'm pretty sure that a wake-up of the IR port of my flowIQ 2200 only is possible by lifting the IR head. I don't think that there is a secret wake-up sequence... I could not find any hint for this. Sad!
> 
> Are there IR heads with a USB controllable electrical magnet? ;-)

You are most likely correct. I am in the process of building a reader (ESP8266 based) that automates this by having a servo-controlled magnet to trigger the meter to wake up. The magnet sensor in the meter seems to be in one particular spot, so only one powerful magnet is needed, but it needs to be placed in the correct location.

Once I have this in place, I can test the max frequency of readout, which I suspect is a handful each day, tops. I also suspect (or perhaps I should say hope) that the meter counts the number of reads, so it might be good to read multiple registers in each read, instead of one read per register.

mdrmdrmdr, 2023-04-30:
Funny, did not think that my idea is taken seriously 😅
Would a electrical magnet not be easier than a construction with a servo? Especially for those already with a IR head? Do you already know the exact required location of the magnet?

copystring, 2023-04-30:
Electro magnets use a lot of power to work correctly and might need to be calibrated?

mroek, 2023-04-30:
I have experimented with an electromagnet, and I'm quite sure it would be possible, but it takes very much power to generate a strong enough field to trigger the sensor. A neodymium magnet is very strong, and the one I have been using is 10x5 mm circular. It is just really simple to use a servo to move the magnet.

I'm doing this as en exercise to develop something interesting with an ESP, but I am still going to push to gain access to the encryption key. If I succeed, this contraption is of course no longer of any use, but that's OK. As I said, I am doing it to educate myself, and I do have all the necessary components lying around anyways.

mroek, 2023-05-02:
I now have a working prototype of my reader. It does use a servo to move a magnet to wake the meter, and then performs a reading before moving the magnet out of the way again. I am currently running tests to find the limits of the polling frequency, but it will take some days until I (hopefully) have an answer.

glance-, 2023-05-02:
There's some magic modes in the meter, evidently by the addon:
https://www.kamstrup.com/en-en/water-solutions/meters-devices/accessories/pulse-adapter

It can put the meter on some mode where it emits a "pulse" over the ir eye which then is turned into a open collector pulse output by the addon. The addon looks to be powered by a saft14500.

mroek, 2023-05-02:
I have also connected the servo directly to a GPIO on the ESP8266, no problem there. The servo is pretty cheap, and it isn't all that precise. Good enough for this, though. I found out that just one 10x5 mm magnet wasn't quite enough, so I stacked two of them to get a stronger field to reliably trigger the meter.

The IR-circuit is also exceedingly simple. I've connected the TX LED to a GPIO through a resistor (the GPIO can sink around 20 mA, which is good enough), and the RX is a NPN phototransistor where the emitter is connected to ground, the collector to a GPIO (I'm using softwareserial, but it could just as well be an UART pin), and with a 1.5 k resistor to VCC (3.3V) to pull it up when there's no light.

Using a phototransistor makes for a simpler circuit, since it is more sensitive than a diode. I used a phototransistor called PT334-6C, which comes in a standard 5 mm package just like a LED.

mroek, 2023-05-02:
> There's some magic modes in the meter, evidently by the addon: https://www.kamstrup.com/en-en/water-solutions/meters-devices/accessories/pulse-adapter
>
> It can put the meter on some mode where it [...]

There's no documentation on that adapter. However, even if we could put the meter into that pulse mode, it still might not stay in that mode forever. It seems that the pulse interface for the flowIQ2200 is different (or it is a newer version):

[image]

The pulse interface uses the same plastic bracket as the normal Kamstrup optical reading head, as you can see (or maybe it is the same, it's just that control box that is different). The meter also stays in the pulse mode for 9 hours after removal of the pulse interface, but there's no telling if it could stay in that mode indefinitely if the interface is left on.

And if it could indeed stay in that mode, since it is a test mode, it might not actually update the normal registers, and it might also stop broadcasting the radio telegrams. That would also be a problem when the municipality does their drive-by to read the meters...

mdrmdrmdr, 2023-05-02:
> I found out that just one 10x5 mm magnet wasn't quite enough, so I stacked two of them to get a stronger field to reliably trigger the meter. 

I have such Neodyms from my last Carrera tuning actions 😊 Can you please share the location the magnet needs to touch the meter?

mroek, 2023-05-02:
> > I found out that just one 10x5 mm magnet wasn't quite enough, so I stacked two of them to get a stronger field to reliably trigger the meter.
> 
> I have such Neodyms from my last Carrera tuning actions 😊 Can you please share the location the magnet needs to touch the meter?

Here's a picture:

[image]

The grey dot indicates the magnet position that I have found to work best. You can easily test this by holding the magnet to that spot for a few seconds. The display should then change mode, and by taking the magnet away and then replacing it, you can cycle the different display views. The orientation of the magnet also matters, for my meter the south pole of the magnet must face the meter. This is opposite of what the 62056-21 standard says, oddly enough.

mdrmdrmdr, 2023-05-02:
Great, thanks a lot. I'm currently not at home, so I can't measure it, but it seems that this position is slightly outside of my 30 mm ø IR head, which would make it easier.

[33 more replies]