esphome/feature-requests #1405: Add support to Yeelight YLKG08Y dimmer Switch

lelemm, 2021-09-10:
**Describe the problem you have/What new integration you would like**
The Yeelight YLKG08Y dimmer switch to be supported by esphome

**Please describe your use case for this integration and alternatives you've tried:**
Use the sensor created from esphome to control other integrations with this dimmer switch.
Tried to use a bluetooth dongle attached to the home assistant, but the signal is too weak. Need something near to get the bluetooth signal (like esphome).

**Additional context**
This is already supported by https://github.com/custom-components/ble_monitor . Probably make the port to esphome easier.
Check this thread: 
https://github.com/nccchirag/yeelight-ble-rotary-dimmer/issues/1
This dimmer switch uses encrypted legacy protocol (old MiBeacon protocol) that must be supported by the esphome in order to work
https://github.com/custom-components/ble_monitor/blob/2404c27d2993599ebe8846b2471d814ee9af368e/custom_components/ble_monitor/ble_parser/xiaomi.py#L717

nagyrobi, 2022-07-07:
Please see https://github.com/dentra/esphome-components/tree/master/components/miot_ylai003, does this work for you?

lelemm, 2022-07-12:
Hi. Tried the component. Does not work. This all it gets logged when used a click or rotary

[15:54:24][D][miot:146]: Got MiBeacon: 58.30.B6.03.13.04.FE.C5.41.24.F8.05.A8.7C.61.65.FE.01.00.00.BD (21)
[15:54:24][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-48 ▂▄▆█
[15:54:25][D][miot:146]: Got MiBeacon: 58.30.B6.03.14.04.FE.C5.41.24.F8.31.E0.D2.E6.3D.0B.01.00.00.A7 (21)
[15:54:25][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-47 ▂▄▆█
[15:54:27][D][miot:146]: Got MiBeacon: 58.30.B6.03.1B.04.FE.C5.41.24.F8.9A.02.1D.2F.2C.F4.01.00.00.ED (21)
[15:54:27][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-58 ▂▄▆█
[15:54:29][D][miot:146]: Got MiBeacon: 58.30.B6.03.1C.04.FE.C5.41.24.F8.1E.1D.01.3F.EB.7A.01.00.00.45 (21)
[15:54:29][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-60 ▂▄▆█
[15:54:30][D][miot:146]: Got MiBeacon: 58.30.B6.03.1C.04.FE.C5.41.24.F8.1E.1D.01.3F.EB.7A.01.00.00.45 (21)
[15:54:30][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-48 ▂▄▆█
[15:54:32][D][miot:146]: Got MiBeacon: 58.30.B6.03.20.04.FE.C5.41.24.F8.88.E3.FD.53.6D.EF.01.00.00.F6 (21)
[15:54:32][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-64 ▂▄▆█
[15:54:32][D][miot:146]: Got MiBeacon: 58.30.B6.03.21.04.FE.C5.41.24.F8.A7.49.BF.75.6F.59.01.00.00.6C (21)
[15:54:32][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-48 ▂▄▆█
[15:54:33][D][miot:146]: Got MiBeacon: 58.30.B6.03.23.04.FE.C5.41.24.F8.2E.6B.CC.0D.C8.A6.01.00.00.9B (21)
[15:54:33][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-52 ▂▄▆█
[15:54:35][D][miot:146]: Got MiBeacon: 58.30.B6.03.28.04.FE.C5.41.24.F8.69.82.72.4B.E8.2A.01.00.00.72 (21)
[15:54:35][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-48 ▂▄▆█
[15:54:49][D][miot:146]: Got MiBeacon: 58.30.B6.03.2D.04.FE.C5.41.24.F8.26.B8.54.12.C9.7D.01.00.00.3D (21)
[15:54:50][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-60 ▂▄▆█
[15:54:50][D][miot:146]: Got MiBeacon: 58.30.B6.03.2E.04.FE.C5.41.24.F8.25.0B.37.AC.07.F8.01.00.00.79 (21)
[15:54:50][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-49 ▂▄▆█
[15:54:51][D][miot:146]: Got MiBeacon: 58.30.B6.03.31.04.FE.C5.41.24.F8.12.54.0C.0C.C1.25.01.00.00.65 (21)
[15:54:51][D][miot:149]:    [03B6] (encrypted) F8:24:41:C5:FE:04 RSSI=-61 ▂▄▆█

lelemm, 2022-07-13:
This one
https://github.com/dentra/esphome-components/tree/master/components/miot_ylkg0xyl
works!

closed 2022-07-14 by nagyrobi